gnuradio/volk #18: volk_profile and make test segfault on volk_32f_s32f_32f_fm_detect_32f kernel.

drmpeg, 2015-04-16:
Function parameters 2 (const float bound) and 3 (float* saveValue) in volk_32f_s32f_32f_fm_detect_32f are reversed. This causes the saveValue pointer to be invalid and the reason for the segfault.

Since the saveValue pointer gets a floating point value, it can often be a reasonable address and this bug can be masked on many systems.

n-west, 2015-04-26:
I've never observed a segfault, but I guess I can see it happening. I think the easiest thing is to make a puppet for this block. I'll get around to it at some point if you don't beat me to it.

wkazubski, 2015-05-02:
This probably makes all my attempts of creating i586 (32-bit) binaries to fail. Build crashes at make test step. If I omit the test, an attempt to run gnuradio-companion fails with "Memory protection error" message.

n-west, 2015-05-03:
Gnuradio-companion is entirely unrelated to VOLK. The qa failure and volk profiler are the only affected applications.

wkazubski, 2015-05-03:
Thanks for advice. Gnuradio-companion crashes too soon to be an error somewhere in volk library.
BTW: If the patch is available, I can test it.

n-west, 2015-05-04:
Here's a proposed fix: https://github.com/n-west/volk/commit/a1902852b5163a03c34b559af2afb2a392196656

If you test it let me know if it fixes your problem and I'll merge it in.

wkazubski, 2015-05-04:
make test still fails on 32-bit intel, but with another error. Gdb says:
gdb test_all
[...]
(gdb) run
[...]/BUILD/volk/build/lib/test_all: symbol lookup error: /home/wk/rpmbuild/BUILD/volk/build/lib/test_all: undefined symbol: volk_32f_x2_fm_detectpuppet_32f_get_func_desc
[Inferior 1 (process 9667) exited with code 0177]
(gdb)

n-west, 2015-05-04:
@wkazubski It looks like you didn't run cmake. Part of the cmake step in VOLK is creating that function in some custome templating code. Run cmake, make, make test should make it work out.

wkazubski, 2015-05-07:
The problem happend while building a binary RPM package using rpmbuild command and a .spec file. Such build is from scratch by default, so it is impossible to omit cmake by mistake.
It also fails when build on openSUSE Build Service:
https://build.opensuse.org/package/show/home:wkazubski:test/volk
64-bit build is OK.

n-west, 2015-05-11:
Since we are diagnosing a build issue now can you try running just cmake without the rpmbuild?

LazyDodo, 2015-05-11:
tested a190285 on windows, solves the crash there.

wkazubski, 2015-05-13:
I run cmake (without any options), make, make test on 32-bit system (SSE1 only). The test failed, but the problem is in single test further down:
> ./test_all 
[...]
RUN_VOLK_TESTS: volk_32fc_s32f_magnitude_16i(131071,1)
generic completed in 5.279ms
u_orc completed in 13.825ms
offset 124509 in1: 324 in2: 325
volk_32fc_s32f_magnitude_16i: fail on arch u_orc
Best aligned arch: generic
Best unaligned arch: generic
Failure on volk_32fc_s32f_magnitude_16i
[...]
Kernel QA finished: 1 failures out of 91 tests.
The following kernels failed QA:
    volk_32fc_s32f_magnitude_16i

wkazubski, 2015-05-13:
Runing the above test on 32-bit SSE2 capable machine (Pentium M Dothan) gives two failures in other places:
> ./test_all
[...]
RUN_VOLK_TESTS: volk_32f_x2_fm_detectpuppet_32f(131071,1)
a_sse completed in 0.657ms
generic completed in 1.559ms
offset 509 in1: -0.0334778 in2: -0.0334778
offset 862 in1: -0.0236661 in2: -0.0236661
offset 8172 in1: 0.0481756 in2: 0.0481756
offset 14101 in1: -0.0593278 in2: -0.0593278
offset 18822 in1: -0.0537884 in2: -0.0537884
offset 19216 in1: 0.035855 in2: 0.0358551
offset 19418 in1: -0.0513217 in2: -0.0513217
offset 19488 in1: -0.0424696 in2: -0.0424695
offset 19820 in1: -0.0389279 in2: -0.0389278
offset 21530 in1: -0.0166605 in2: -0.0166605
volk_32f_x2_fm_detectpuppet_32f: fail on arch a_sse
Best aligned arch: a_sse
Best unaligned arch: generic
Failure on volk_32f_x2_fm_detectpuppet_32f
[...]
RUN_VOLK_TESTS: volk_16ic_magnitude_16i(131071,1)
a_sse completed in 2.052ms
generic completed in 4.854ms
offset 3430 in1: -31279 in2: -31280
offset 10499 in1: 30064 in2: 30063
offset 13644 in1: -24050 in2: -24051
offset 14759 in1: -30169 in2: -30168
offset 19084 in1: -28368 in2: -28367
offset 22521 in1: -32762 in2: -32763
offset 23126 in1: -32762 in2: -32763
offset 32435 in1: -24024 in2: -24023
offset 41967 in1: -32762 in2: -32763
offset 42358 in1: -22535 in2: -22536
volk_16ic_magnitude_16i: fail on arch a_sse
Best aligned arch: a_sse
Best unaligned arch: generic
Failure on volk_16ic_magnitude_16i
[...]
Kernel QA finished: 2 failures out of 91 tests.
The following kernels failed QA:
    volk_32f_x2_fm_detectpuppet_32f
    volk_16ic_magnitude_16i

Previous test was on Athlon XP 2400 Thoroughbred

n-west, 2015-05-13:
OK, some progress. The magnitude failures are harmless and I have a fix for it in a dev branch of mine. The fm_detect failure is a little troubling since those values look exactly the same. Those numbers are all really small so perhaps it's just a tolerance issue.

This suggests that your cmake set up via rpmbuild was bad in some way-- are you certain they were fresh builds? Since we no longer segfault

n-west, 2015-05-18:
I'm considering this closed since the tests no longer crash. If they are failing that is a separate issue that also needs to be resolved.

closed 2015-05-18 by n-west