gnuradio/volk #24: Fatal error: volk/constants.h: no such file or directory

p-brane, 2015-06-04:
I'm trying to compile this for gnuradio on the Raspberry Pi.  cmake ../ failed because  DOXYGEN_EXECUTABLE,  ORC_LIBRARY ORC_INCLUDE_DIR ORCC_EXECUTABLE were missing plus cmake error CMake >= 2.8.10 is required for ARM NEON support occurred. 

I found someone on reddit suggested

cmake -Dhave_mfpu_neon=0 -DCMAKE_CXX_FLAGS:STRING="-march=armv6  -mfpu=vfp -mfloat-abi=hard" -DCMAKE_C_FLAGS:STRING="-march=armv6 -mfpu=vfp -mfloat-abi=hard" ../

This created build files but still some warning occurred

-- Not enabling ASM support. CMake >= 2.8.10 required.
-- Did not find liborc and orcc, disabling orc support...
-- Loading build date Wed, 03 Jun 2015 21:37:46 into constants...
-- Loading version v1.0-32-g372ad7dc into constants...
-- Using install prefix: /usr/local
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/volk/build

Make then produce the following errors

[ 40%] Building C object lib/CMakeFiles/volk_obj.dir/constants.c.o
/home/pi/volk/build/lib/constants.c:27:28: fatal error: volk/constants.h: No such file or directory
compilation terminated.
lib/CMakeFiles/volk_obj.dir/build.make:1143: recipe for target 'lib/CMakeFiles/volk_obj.dir/constants.c.o' failed
make[2]: *** [lib/CMakeFiles/volk_obj.dir/constants.c.o] Error 1
CMakeFiles/Makefile2:180: recipe for target 'lib/CMakeFiles/volk_obj.dir/all' failed
make[1]: *** [lib/CMakeFiles/volk_obj.dir/all] Error 2
Makefile:123: recipe for target 'all' failed
make: *** [all] Error 2

Would you provide some direction on how to correct this error?

douggeiger, 2015-06-05:
This does not appear to be a bug in Volk. I would recommend trying the [mailing list](http://gnuradio.org/redmine/projects/gnuradio/wiki/MailingLists) to get assistance with this problem, and in the meantime you should probably read through this [page](http://gnuradio.org/redmine/projects/gnuradio/wiki/Embedded), and in particular the section on [native compiling](http://gnuradio.org/redmine/projects/gnuradio/wiki/Embedded#Native-Compiling).

n-west, 2015-06-05:
@p-brane I agree with Doug here. This looks and feels like a non-bug. Unfortunately I haven't set up a VOLK specific mailing list to discuss things like this, but this is perfectly in the scope of discuss-gnuradio.

It sounds like you've tried a couple of things and keep getting marginally closer. Make sure you delete your build directory and start from scratch with the correct C_FLAGS and CXX_FLAGS if you're doing a native ARM build. If your still stuck send your cmake output; it might be helpful. Preferably this discussion can continue on discuss-gnuradio where you're more likely to find someone else who has compiled VOLK+gnuradio successfully on a raspberry pi. (I *think* these issues are primarily seen by me and Doug and neither of us use a raspberry pi or armv6.

n-west, 2015-07-09:
Closing because we haven't heard from this in a while and I don't think it's a VOLK issue anyway.

closed 2015-07-09 by n-west