gnuradio/volk #27: problem compile gnuradio

tele1, 2015-07-07:
PCLinuxOS 64bit

I used:

%cmake
%make
%makeinstall_std

compiled volk 100% succes

problem is, gnu radio can not find volk
from gnu radio cmake I see, this search volk_common.h
volk_common.h is installed in
/usr/local/include/volk/volk_common.h
 
how fix it ?

macros :
$ rpm --eval=%cmake

    
  CFLAGS="${CFLAGS:--O2  -pipe -Wformat -Wp,-D_FORTIFY_SOURCE=2 -fstack-protector --param=ssp-buffer-size=4}" ; export CFLAGS ; 
  CXXFLAGS="${CXXFLAGS:--O2  -pipe -Wformat -Wp,-D_FORTIFY_SOURCE=2 -fstack-protector --param=ssp-buffer-size=4}" ; export CXXFLAGS ; 
  FFLAGS="${FFLAGS:--O2  -pipe -Wformat -Wp,-D_FORTIFY_SOURCE=2 -fstack-protector --param=ssp-buffer-size=4}" ; export FFLAGS ; 
  LDFLAGS="$LDFLAGS -Wl,--as-needed -Wl,--no-undefined -Wl,-z,relro -Wl,-O1 -Wl,--build-id"; export LDFLAGS ; 
    mkdir -p build 
    cd build 
         
    /usr/bin/cmake .. \
        -DCMAKE_INSTALL_PREFIX:PATH=/usr \
        -DCMAKE_INSTALL_LIBDIR:PATH=/usr/lib64 \
        -DCMAKE_INSTALL_LIBEXECDIR:PATH=/usr/lib64 \
        -DCMAKE_INSTALL_SYSCONFDIR:PATH=/etc \
        -DINCLUDE_INSTALL_DIR:PATH=/usr/include \
        -DLIB_INSTALL_DIR:PATH=/usr/lib64 \
        -DSYSCONF_INSTALL_DIR:PATH=/etc \
        -DSHARE_INSTALL_PREFIX:PATH=/usr/share \
        -DCMAKE_BUILD_TYPE=release \
%if "lib64" == "lib64" 
        -DLIB_SUFFIX=64 \
%endif 
        -DCMAKE_SKIP_RPATH:BOOL=ON \
         \
        -DBUILD_SHARED_LIBS:BOOL=ON \
        -DBUILD_STATIC_LIBS:BOOL=OFF \
        -DCMAKE_MODULE_LINKER_FLAGS="-Wl,--as-needed  -Wl,-z,relro -Wl,-O1 -Wl,--build-id"

$ rpm --eval=%make
make -O -j4

$ rpm --eval=%makeinstall_std
make DESTDIR=/home/gg/src/rpm/BUILDROOT/%{name}-%{version}-%{release}.x86_64 install

closed 2015-07-07 by tele1

n-west, 2015-07-07:
In the future, this is not the correct forum for installation help. This is for reporting issues related to the software. For help email the GNU Radio mailing list.

Also FYI: there's multiple ways to install VOLK + GNU Radio. I think you should have seen a message output from cmake instructing you to checkout a submodule. Alternatively since you had VOLK installed already you could have used the cmake option USE_INTERNAL_VOLK=Off

tele1, 2015-07-07:
Thanks,
now I see inside gnuradio volk, so I try disable volk :)