gnuradio/volk #43: volk_32f_x3_sum_of_poly_32f fails on OSX Yosemite

bhilburn, 2015-08-29:
Noticed it today, confirmed by @michaelld.

Using d063b05c7e7c.

```
Darwin 14.5.0 Darwin Kernel Version 14.5.0
root:xnu-2782.40.9~1/RELEASE_X86_64 x86_64
```

```
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
Apple LLVM version 6.1.0 (clang-602.0.53) (based on LLVM 3.6.0svn)
Target: x86_64-apple-darwin14.5.0
Thread model: posix
```

michaelld, 2015-08-29:
Yes, I also see this; my setup is Mac OS X 10.9 (Darwin Kernel Version 13.4.0, x86_64). I'll also test on 10.8 soonish, and will be able to test on 10.11b3 sometime next week if all goes as planned / hoped.

n-west, 2015-09-16:
Which protokernel is failing? SSE3, AVX, or both?

michaelld, 2015-09-17:
Now that the port volk-devel works again, here's the output from this test:
```
1: RUN_VOLK_TESTS: volk_32f_x3_sum_of_poly_32f(131071,1)
1: a_sse3 completed in 0.085ms
1: a_avx completed in 0.088ms
1: generic completed in 0.321ms
1: u_avx completed in 0.078ms
1: offset 0 in1: -19656.1 in2: -19689.5
1: volk_32f_x3_sum_of_poly_32f: fail on arch a_sse3
1: offset 0 in1: -19656.1 in2: -19686
1: volk_32f_x3_sum_of_poly_32f: fail on arch a_avx
1: offset 0 in1: -19656.1 in2: -19686
1: volk_32f_x3_sum_of_poly_32f: fail on arch u_avx
1: Best aligned arch: u_avx
1: Best unaligned arch: u_avx
1: Failure on volk_32f_x3_sum_of_poly_32f
```

n-west, 2015-10-06:
Hmm. Weird that it's only showing up on Apple machines since it looks close enough that it might just be a tolerance issue

michaelld, 2015-10-08:
Same results from 10.11 32-bit:
```
1: RUN_VOLK_TESTS: volk_32f_x3_sum_of_poly_32f(131071,1)
1: a_sse3 completed in 0.094ms
1: a_avx completed in 0.066ms
1: generic completed in 0.279ms
1: u_avx completed in 0.06ms
1: offset 0 in1: -19656.1 in2: -19689.5
1: volk_32f_x3_sum_of_poly_32f: fail on arch a_sse3
1: offset 0 in1: -19656.1 in2: -19686
1: volk_32f_x3_sum_of_poly_32f: fail on arch a_avx
1: offset 0 in1: -19656.1 in2: -19686
1: volk_32f_x3_sum_of_poly_32f: fail on arch u_avx
1: Best aligned arch: u_avx
1: Best unaligned arch: u_avx
1: Failure on volk_32f_x3_sum_of_poly_32f
```
and 10.11 64-bit:
```
1: RUN_VOLK_TESTS: volk_32f_x3_sum_of_poly_32f(131071,1)
1: a_sse3 completed in 0.084ms
1: a_avx completed in 0.075ms
1: generic completed in 0.278ms
1: u_avx completed in 0.06ms
1: offset 0 in1: -19656.1 in2: -19689.5
1: volk_32f_x3_sum_of_poly_32f: fail on arch a_sse3
1: offset 0 in1: -19656.1 in2: -19686
1: volk_32f_x3_sum_of_poly_32f: fail on arch a_avx
1: offset 0 in1: -19656.1 in2: -19686
1: volk_32f_x3_sum_of_poly_32f: fail on arch u_avx
1: Best aligned arch: u_avx
1: Best unaligned arch: u_avx
1: Failure on volk_32f_x3_sum_of_poly_32f
```

michaelld, 2015-10-08:
Seems like it might be a rounding error?

gnieboer, 2016-03-16:
Found the same issue on MSVC Windows 64-bit, but only on a_sse3.

volk_32f_x3_sum_of_poly_32f ... failed in arch a_sse3 ... offset 0 in1: 32611.7 in2: 32554.9 tolerance was: 0.001

n-west, 2017-02-24:
🤔

michaelld, 2017-02-24:
Been a long time. Probably fixed by now.

michaelld, 2017-04-03:
With PR #112 in place, this issue is no longer relevant on Mac OS X 10.7 / 8 & 10.12 (latest) -- meaning that Volk passes "make test" cleanly. So, I think it can be closed if it's not already.

closed 2017-08-14 by n-west