gnuradio/volk #44: _mm_bsrli_si128 requires Clang 3.7+

michaelld, 2015-08-29:
volk_8u_x2_encodeframepolar_8u uses _mm_bsrli_si128, which requires specific recent compiler versions (e.g., Clang 3.7+). Thus, we need to check for these versions in the CMake configuration script, which is not done at least for Clang right now. I'll test on GCC to see if its done there, next week.

michaelld, 2015-08-29:
Looks like this intrinsic is not part of GCC 4.7, but it is part of GCC 4.8. It is not part of Clang 3.6, but it is in Clang 3.7. So, to use these kernels we need to be checking for compiler version & not building or using them if the compiler is too old.

n-west, 2015-09-02:
I'm aware of this and just haven't had time to think about how to handle it. I'll probably be able to get to it tomorrow evening along with the PR queue.

Note to anyone interested: Currently we glob to get all of the headers in kernels/volk/*.h. This might require changing that (lib/CMakeLists.txt:322) so we can include the new polar stuff conditionally based on the compiler version. The alternative is to add a new arch for this intrinsic which is pretty ugly.

michaelld, 2015-09-03:
OK; glad you're aware of it. Taking a look at the CMakeLists.txt where you say, I'll let you take the lead on this. I'm happy to do testing & some tweaks, of course.

n-west, 2015-09-16:
I thought I might come up with something while this sat in the back of my mind while I recovered from grcon and caught up with other things. I think I'm mostly caught up with the rest of VOLK, but not much has come to mind on this issue. Any more insights on compiler versions?

michaelld, 2015-09-16:
Yes. Testing this week (maybe even today). If it works I'll do a PR & note that here.

michaelld, 2015-09-17:
Per discussion on IRC freenode #volk, looks like we can use _mm_srli_si128 instead of _mm_bsrli_si128. This intrinsic goes all the way back to LLVM 3.3 or earlier; GCC 4.2 or earlier. Very nice!

michaelld, 2015-09-17:
replacing "s/_mm_bsrli_si128/_mm_srli_si128/g" works for me here, no special compiler checking needed!

michaelld, 2015-09-18:
Do you want me to do a PR with this change? Seems simple enough, but it might save you a few steps.

n-west, 2015-09-21:
Yea, you can do a PR or I'll get around to when I get home tonight.

michaelld, 2015-09-21:
PR issued.

closed 2015-10-06 by n-west