gnuradio/volk #45: Volk fail to build on CentOS 6.4 / 6.5 / 6.6

sgardner340, 2015-09-01:
21:35:41 Current step: (volk :: make)
21:35:41 make
21:35:41 bashexec (/home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.5_Desktop_x86_VM_R/satisfy_order/rpm-src/src/volk/build):: 
21:35:41     find . -name '*.make' -print0 | xargs -0 sed -i 's/\(^.*\)\(--cyan \"\)\(.*\)\"$/\1\2\x1b[36;1m\3\x1b[0m\"/;s/\(^.*\)\(--green \"\)\(.*\)\"$/\1\2\x1b[32;1m\3\x1b[0m\"/;s/\(^.*\)\(--red --bold \"\)\(.*\)\"$/\1\2\x1b[31;1m\3\x1b[0m\"/;s/\(^.*\)\(--blue --bold \"\)\(.*\)\"$/\1\2\x1b[34;1m\3\x1b[0m\"/'
21:35:41     find . -name 'Makefile' -print0 | xargs -0 sed -i 's/\(^.*\)\(--cyan \"\)\(.*\)\"$/\1\2\x1b[36;1m\3\x1b[0m\"/;s/\(^.*\)\(--green \"\)\(.*\)\"$/\1\2\x1b[32;1m\3\x1b[0m\"/;s/\(^.*\)\(--red --bold \"\)\(.*\)\"$/\1\2\x1b[31;1m\3\x1b[0m\"/;s/\(^.*\)\(--blue --bold \"\)\(.*\)\"$/\1\2\x1b[34;1m\3\x1b[0m\"/'
21:35:41     ln -s /home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.5_Desktop_x86_VM_R/satisfy_order/rpm-src/colorgcc.pl /apps/smokey64/bin/cc
21:35:41     ln -s /home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.5_Desktop_x86_VM_R/satisfy_order/rpm-src/colorgcc.pl /apps/smokey64/bin/gcc
21:35:41     ln -s /home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.5_Desktop_x86_VM_R/satisfy_order/rpm-src/colorgcc.pl /apps/smokey64/bin/c++
21:35:41     ln -s /home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.5_Desktop_x86_VM_R/satisfy_order/rpm-src/colorgcc.pl /apps/smokey64/bin/g++
21:35:41     make -j4
21:35:41 
21:35:41 Scanning dependencies of target pygen_python_volk_modtool_1a4b4
21:35:41 [  2%] [  5%] [  7%] Scanning dependencies of target pygen_python_volk_modtool_760e7
21:35:41 Shebangin volk_modtool
21:35:41 Generating ../include/volk/volk.h
21:35:41 [ 10%] Generating volk_machine_avx_64_mmx.c
21:35:41 Generating __init__.pyc, cfg.pyc, volk_modtool_generate.pyc
21:35:41 [ 10%] Built target pygen_python_volk_modtool_1a4b4
21:35:41 [ 12%] Generating __init__.pyo, cfg.pyo, volk_modtool_generate.pyo
21:35:41 [ 12%] Built target pygen_python_volk_modtool_760e7
21:35:41 [ 15%] [ 17%] Generating volk.c
21:35:41 Generating ../include/volk/volk_typedefs.h
21:35:42 /apps/smokey64/lib64/python2.6/site-packages/Cheetah/CacheRegion.py:30: DeprecationWarning: the md5 module is deprecated; use hashlib instead
21:35:42   import md5
21:35:42 /apps/smokey64/lib64/python2.6/site-packages/Cheetah/CacheRegion.py:30: DeprecationWarning: the md5 module is deprecated; use hashlib instead
21:35:42   import md5
21:35:42 /apps/smokey64/lib64/python2.6/site-packages/Cheetah/CacheRegion.py:30: DeprecationWarning: the md5 module is deprecated; use hashlib instead
21:35:42   import md5
21:35:42 [ 20%] Generating ../include/volk/volk_cpu.h
21:35:42 [ 23%] Generating volk_cpu.c
21:35:42 [ 25%] Generating ../include/volk/volk_config_fixed.h
21:35:42 /apps/smokey64/lib64/python2.6/site-packages/Cheetah/CacheRegion.py:30: DeprecationWarning: the md5 module is deprecated; use hashlib instead
21:35:42   import md5
21:35:42 [ 28%] Generating volk_machines.h
21:35:42 /apps/smokey64/lib64/python2.6/site-packages/Cheetah/CacheRegion.py:30: DeprecationWarning: the md5 module is deprecated; use hashlib instead
21:35:42   import md5
21:35:42 [ 30%] Generating volk_machines.c
21:35:42 /apps/smokey64/lib64/python2.6/site-packages/Cheetah/CacheRegion.py:30: DeprecationWarning: the md5 module is deprecated; use hashlib instead
21:35:42   import md5
21:35:42 [ 33%] /apps/smokey64/lib64/python2.6/site-packages/Cheetah/CacheRegion.py:30: DeprecationWarning: the md5 module is deprecated; use hashlib instead
21:35:42   import md5
21:35:42 Generating volk_machine_generic.c
21:35:42 [ 35%] Generating volk_machine_sse2_64_mmx.c
21:35:43 /apps/smokey64/lib64/python2.6/site-packages/Cheetah/CacheRegion.py:30: DeprecationWarning: the md5 module is deprecated; use hashlib instead
21:35:43   import md5
21:35:43 /apps/smokey64/lib64/python2.6/site-packages/Cheetah/CacheRegion.py:30: DeprecationWarning: the md5 module is deprecated; use hashlib instead
21:35:43   import md5
21:35:43 /apps/smokey64/lib64/python2.6/site-packages/Cheetah/CacheRegion.py:30: DeprecationWarning: the md5 module is deprecated; use hashlib instead
21:35:43   import md5
21:35:43 [ 38%] Generating volk_machine_sse3_64.c
21:35:43 [ 41%] Generating volk_machine_ssse3_64.c
21:35:43 [ 43%] Generating volk_machine_sse4_a_64.c
21:35:43 /apps/smokey64/lib64/python2.6/site-packages/Cheetah/CacheRegion.py:30: DeprecationWarning: the md5 module is deprecated; use hashlib instead
21:35:43   import md5
21:35:43 [ 46%] Generating volk_machine_sse4_1_64.c
21:35:43 /apps/smokey64/lib64/python2.6/site-packages/Cheetah/CacheRegion.py:30: DeprecationWarning: the md5 module is deprecated; use hashlib instead
21:35:43   import md5
21:35:43 /apps/smokey64/lib64/python2.6/site-packages/Cheetah/CacheRegion.py:30: DeprecationWarning: the md5 module is deprecated; use hashlib instead
21:35:43   import md5
21:35:43 [ 48%] Generating volk_machine_sse4_2_64.c
21:35:43 /apps/smokey64/lib64/python2.6/site-packages/Cheetah/CacheRegion.py:30: DeprecationWarning: the md5 module is deprecated; use hashlib instead
21:35:43   import md5
21:35:43 /apps/smokey64/lib64/python2.6/site-packages/Cheetah/CacheRegion.py:30: DeprecationWarning: the md5 module is deprecated; use hashlib instead
21:35:43   import md5
21:35:44 /apps/smokey64/lib64/python2.6/site-packages/Cheetah/CacheRegion.py:30: DeprecationWarning: the md5 module is deprecated; use hashlib instead
21:35:44   import md5
21:35:44 Scanning dependencies of target volk_obj
21:35:44 [ 51%] [ 53%] [ 56%] [ 58%] Building C object lib/CMakeFiles/volk_obj.dir/constants.c.o
21:35:44 Building C object lib/CMakeFiles/volk_obj.dir/volk_prefs.c.o
21:35:44 Building C object lib/CMakeFiles/volk_obj.dir/volk_malloc.c.o
[... 26 lines cut ...]
21:35:45 make[2]: *** Waiting for unfinished jobs....
21:35:47 make[1]: *** [lib/CMakeFiles/volk_obj.dir/all] Error 2
21:35:47 make: *** [all] Error 2
21:35:47 bash return val = 2
21:35:47 Traceback (most recent call last):
21:35:47   File "./pybombs", line 360, in <module>
21:35:47     main()
21:35:47   File "./pybombs", line 357, in main
21:35:47     sys.exit(cli(parser,opts,args))
21:35:47   File "./pybombs", line 171, in cli
21:35:47     pybombs_ops.install(p, not opts.force);
21:35:47   File "/home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.5_Desktop_x86_VM_R/satisfy_order/rpm-src/mod_pybombs/pybombs_ops.py", line 151, in install
21:35:47     global_recipes[pkg].install();
21:35:47   File "/home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.5_Desktop_x86_VM_R/satisfy_order/rpm-src/mod_pybombs/recipe.py", line 531, in install
21:35:47     st = self.install_src();
21:35:47   File "/home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.5_Desktop_x86_VM_R/satisfy_order/rpm-src/mod_pybombs/recipe.py", line 598, in install_src
21:35:47     self.install_order[step][1]();
21:35:47   File "/home/hudson/workspace/PyBombs_Build_All_Matrix/os/CentOS_6.5_Desktop_x86_VM_R/satisfy_order/rpm-src/mod_pybombs/recipe.py", line 658, in make
21:35:47     assert(st == 0);
21:35:47 AssertionError
21:35:47 Build step 'Execute shell' marked build as failure
21:35:47 [WARNINGS] Skipping publisher since build result is FAILURE
21:35:47 Email was triggered for: Still Failing
21:35:47 Trigger Failure was overridden by another trigger and will not send an email.
21:35:47 Sending email for trigger: Still Failing

n-west, 2015-09-01:
Thanks for the report. In the future please paste make output with only 1 job so it's easier to read. This appears to be a duplicate of #44. Can you tell me your compiler version(s)?

sgardner340, 2015-09-01:
CentOS 6.4: gcc version 4.4.7 20120313 (Red Hat 4.4.7-11) (GCC)
CentOS 6.5: gcc version 4.4.7 20120313 (Red Hat 4.4.7-11) (GCC)
CentOS 6.6: gcc version 4.4.7 20120313 (Red Hat 4.4.7-11) (GCC)

n-west, 2015-10-06:
Should be fixed now. Holler if it's not.

closed 2015-10-06 by n-west

sgardner340, 2015-10-07:
Volk master works fine for me​