gnuradio/volk #46: Volk (master) failed to build on Ubuntu 12.04 / 12.10 / 13.04

sgardner340, 2015-09-01:
21:11:50 Current step: (volk :: make)
21:11:50 make
21:11:50 bashexec (/home/hudson/workspace/PyBombs_Build_All_Matrix/os/Ubuntu_13.04_Desktop_x86_64_VM/satisfy_order/deb-src/src/volk/build):: 
21:11:50     find . -name '*.make' -print0 | xargs -0 sed -i 's/\(^.*\)\(--cyan \"\)\(.*\)\"$/\1\2\x1b[36;1m\3\x1b[0m\"/;s/\(^.*\)\(--green \"\)\(.*\)\"$/\1\2\x1b[32;1m\3\x1b[0m\"/;s/\(^.*\)\(--red --bold \"\)\(.*\)\"$/\1\2\x1b[31;1m\3\x1b[0m\"/;s/\(^.*\)\(--blue --bold \"\)\(.*\)\"$/\1\2\x1b[34;1m\3\x1b[0m\"/'
21:11:50     find . -name 'Makefile' -print0 | xargs -0 sed -i 's/\(^.*\)\(--cyan \"\)\(.*\)\"$/\1\2\x1b[36;1m\3\x1b[0m\"/;s/\(^.*\)\(--green \"\)\(.*\)\"$/\1\2\x1b[32;1m\3\x1b[0m\"/;s/\(^.*\)\(--red --bold \"\)\(.*\)\"$/\1\2\x1b[31;1m\3\x1b[0m\"/;s/\(^.*\)\(--blue --bold \"\)\(.*\)\"$/\1\2\x1b[34;1m\3\x1b[0m\"/'
21:11:50     ln -s /home/hudson/workspace/PyBombs_Build_All_Matrix/os/Ubuntu_13.04_Desktop_x86_64_VM/satisfy_order/deb-src/colorgcc.pl /apps/smokey64/bin/cc
21:11:50     ln -s /home/hudson/workspace/PyBombs_Build_All_Matrix/os/Ubuntu_13.04_Desktop_x86_64_VM/satisfy_order/deb-src/colorgcc.pl /apps/smokey64/bin/gcc
21:11:50     ln -s /home/hudson/workspace/PyBombs_Build_All_Matrix/os/Ubuntu_13.04_Desktop_x86_64_VM/satisfy_order/deb-src/colorgcc.pl /apps/smokey64/bin/c++
21:11:50     ln -s /home/hudson/workspace/PyBombs_Build_All_Matrix/os/Ubuntu_13.04_Desktop_x86_64_VM/satisfy_order/deb-src/colorgcc.pl /apps/smokey64/bin/g++
21:11:50     make -j4
21:11:50 
21:11:50 Scanning dependencies of target pygen_python_volk_modtool_e43da
21:11:50 Scanning dependencies of target pygen_python_volk_modtool_67216
21:11:50 [  4%] [  4%] [  7%] [  9%] Generating ../include/volk/volk.h
21:11:50 Shebangin volk_modtool
21:11:50 Generating volk_machine_avx2_64_mmx.c
21:11:50 Generating __init__.pyc, cfg.pyc, volk_modtool_generate.pyc
21:11:50 [  9%] Built target pygen_python_volk_modtool_e43da
21:11:50 [ 12%] [ 14%] Generating volk.c
21:11:50 Generating __init__.pyo, cfg.pyo, volk_modtool_generate.pyo
21:11:50 [ 14%] Built target pygen_python_volk_modtool_67216
21:11:50 [ 17%] Generating ../include/volk/volk_typedefs.h
21:11:50 [ 19%] Generating ../include/volk/volk_cpu.h
21:11:50 [ 21%] Generating volk_cpu.c
21:11:50 [ 24%] Generating ../include/volk/volk_config_fixed.h
21:11:50 [ 26%] Generating volk_machines.h
21:11:51 [ 29%] Generating volk_machines.c
21:11:51 [ 31%] Generating volk_machine_generic.c
21:11:51 [ 34%] Generating volk_machine_sse2_64_mmx.c
21:11:51 [ 36%] Generating volk_machine_sse3_64.c
21:11:51 [ 39%] Generating volk_machine_ssse3_64.c
21:11:51 [ 41%] Generating volk_machine_sse4_a_64.c
21:11:51 [ 43%] Generating volk_machine_sse4_1_64.c
21:11:51 [ 46%] Generating volk_machine_sse4_2_64.c
21:11:51 [ 48%] Generating volk_machine_avx_64_mmx.c
21:11:52 Scanning dependencies of target volk_obj
21:11:52 [ 53%] [ 53%] [ 58%] [ 58%] Building C object lib/CMakeFiles/volk_obj.dir/constants.c.o
21:11:52 Building C object lib/CMakeFiles/volk_obj.dir/volk_prefs.c.o
21:11:52 Building C object lib/CMakeFiles/volk_obj.dir/volk_malloc.c.o
21:11:52 Building C object lib/CMakeFiles/volk_obj.dir/volk_rank_archs.c.o
21:11:52 [ 60%] Building C object lib/CMakeFiles/volk_obj.dir/volk.c.o
21:11:52 [ 63%] Building C object lib/CMakeFiles/volk_obj.dir/volk_cpu.c.o
21:11:52 [ 65%] Building C object lib/CMakeFiles/volk_obj.dir/volk_machines.c.o
21:11:52 [ 68%] Building C object lib/CMakeFiles/volk_obj.dir/volk_machine_generic.c.o
21:11:52 [ 70%] Building C object lib/CMakeFiles/volk_obj.dir/volk_machine_sse2_64_mmx.c.o
21:11:52 [ 73%] Building C object lib/CMakeFiles/volk_obj.dir/volk_machine_sse3_64.c.o
21:11:54 [ 75%] Building C object lib/CMakeFiles/volk_obj.dir/volk_machine_ssse3_64.c.o
21:11:55 In file included from /home/hudson/workspace/PyBombs_Build_All_Matrix/os/Ubuntu_13.04_Desktop_x86_64_VM/satisfy_order/deb-src/src/volk/kernels/volk/volk_8u_x3_encodepolar_8u_x2.h:78:0,
21:11:55                  from /home/hudson/workspace/PyBombs_Build_All_Matrix/os/Ubuntu_13.04_Desktop_x86_64_VM/satisfy_order/deb-src/src/volk/kernels/volk/volk_8u_x3_encodepolarpuppet_8u.h:30,:
21:11:55                  from /home/hudson/workspace/PyBombs_Build_All_Matrix/os/Ubuntu_13.04_Desktop_x86_64_VM/satisfy_order/deb-src/src/volk/build/lib/volk_machine_ssse3_64.c:100:
21:11:55 /home/hudson/workspace/PyBombs_Build_All_Matrix/os/Ubuntu_13.04_Desktop_x86_64_VM/satisfy_order/deb-src/src/volk/kernels/volk/volk_8u_x2_encodeframepolar_8u.h: In function ‘volk_8u_x2_encodeframepolar_8u_u_ssse3’:
21:11:55 /home/hudson/workspace/PyBombs_Build_All_Matrix/os/Ubuntu_13.04_Desktop_x86_64_VM/satisfy_order/deb-src/src/volk/kernels/volk/volk_8u_x2_encodeframepolar_8u.h:124:11: warning: implicit declaration of function ‘_mm_bsrli_si128’ [-Wimplicit-function-declaration]
21:11:55            shifted = _mm_bsrli_si128(r_temp0, 1);
21:11:55            ^
21:11:55 /home/hudson/workspace/PyBombs_Build_All_Matrix/os/Ubuntu_13.04_Desktop_x86_64_VM/satisfy_order/deb-src/src/volk/kernels/volk/volk_8u_x2_encodeframepolar_8u.h:124:19: error: incompatible types when assigning to type ‘__m128i’ from type ‘int’
21:11:55            shifted = _mm_bsrli_si128(r_temp0, 1);
21:11:55                    ^
21:11:55 /home/hudson/workspace/PyBombs_Build_All_Matrix/os/Ubuntu_13.04_Desktop_x86_64_VM/satisfy_order/deb-src/src/volk/kernels/volk/volk_8u_x2_encodeframepolar_8u.h:129:19: error: incompatible types when assigning to type ‘__m128i’ from type ‘int’
21:11:55            shifted = _mm_bsrli_si128(r_temp1, 1);
21:11:55                    ^
21:11:55 /home/hudson/workspace/PyBombs_Build_All_Matrix/os/Ubuntu_13.04_Desktop_x86_64_VM/satisfy_order/deb-src/src/volk/kernels/volk/volk_8u_x2_encodeframepolar_8u.h:173:13: error: incompatible types when assigning to type ‘__m128i’ from type ‘int’
21:11:55      shifted = _mm_bsrli_si128(r_temp0, 8);
21:11:55              ^
21:11:55 /home/hudson/workspace/PyBombs_Build_All_Matrix/os/Ubuntu_13.04_Desktop_x86_64_VM/satisfy_order/deb-src/src/volk/kernels/volk/volk_8u_x2_encodeframepolar_8u.h:181:13: error: incompatible types when assigning to type ‘__m128i’ from type ‘int’
21:11:55      shifted = _mm_bsrli_si128(r_frame0, 4);
21:11:55              ^
21:11:55 /home/hudson/workspace/PyBombs_Build_All_Matrix/os/Ubuntu_13.04_Desktop_x86_64_VM/satisfy_order/deb-src/src/volk/kernels/volk/volk_8u_x2_encodeframepolar_8u.h:185:13: error: incompatible types when assigning to type ‘__m128i’ from type ‘int’
21:11:55      shifted = _mm_bsrli_si128(r_frame0, 2);
21:11:55              ^
21:11:55 /home/hudson/workspace/PyBombs_Build_All_Matrix/os/Ubuntu_13.04_Desktop_x86_64_VM/satisfy_order/deb-src/src/volk/kernels/volk/volk_8u_x2_encodeframepolar_8u.h:189:13: error: incompatible types when assigning to type ‘__m128i’ from type ‘int’
[... 21 lines cut ...]
21:11:55 make[2]: *** [lib/CMakeFiles/volk_obj.dir/volk_machine_ssse3_64.c.o] Error 1
21:11:55 make[2]: *** Waiting for unfinished jobs....
21:11:58 make[1]: *** [lib/CMakeFiles/volk_obj.dir/all] Error 2
21:11:58 make: *** [all] Error 2
21:11:58 bash return val = 2
21:11:58 Traceback (most recent call last):
21:11:58   File "./pybombs", line 360, in <module>
21:11:58     main()
21:11:58   File "./pybombs", line 357, in main
21:11:58     sys.exit(cli(parser,opts,args))
21:11:58   File "./pybombs", line 171, in cli
21:11:58     pybombs_ops.install(p, not opts.force);
21:11:58   File "/home/hudson/workspace/PyBombs_Build_All_Matrix/os/Ubuntu_13.04_Desktop_x86_64_VM/satisfy_order/deb-src/mod_pybombs/pybombs_ops.py", line 151, in install
21:11:58     global_recipes[pkg].install();
21:11:58   File "/home/hudson/workspace/PyBombs_Build_All_Matrix/os/Ubuntu_13.04_Desktop_x86_64_VM/satisfy_order/deb-src/mod_pybombs/recipe.py", line 531, in install
21:11:58     st = self.install_src();
21:11:58   File "/home/hudson/workspace/PyBombs_Build_All_Matrix/os/Ubuntu_13.04_Desktop_x86_64_VM/satisfy_order/deb-src/mod_pybombs/recipe.py", line 598, in install_src
21:11:58     self.install_order[step][1]();
21:11:58   File "/home/hudson/workspace/PyBombs_Build_All_Matrix/os/Ubuntu_13.04_Desktop_x86_64_VM/satisfy_order/deb-src/mod_pybombs/recipe.py", line 658, in make
21:11:58     assert(st == 0);
21:11:58 AssertionError
21:11:58 Build step 'Execute shell' marked build as failure
21:11:58 [WARNINGS] Skipping publisher since build result is FAILURE
21:11:58 Email was triggered for: Still Failing
21:11:58 Trigger Failure was overridden by another trigger and will not send an email.

n-west, 2015-09-01:
This is a duplicate of #45. Is there a reason to have a second issue open?

closed 2015-09-01 by n-west

sgardner340, 2015-09-01:
This issue can be closed as long as the #45 fix solves the CentOS as well
as the Ubuntu issue.