gnuradio/volk #51: Compilation failure

staticfloat, 2015-10-08:
On a fresh clone on OSX 10.10:

```
In file included from /Users/sabae/src/gnuradio/gnuradio/volk/build/lib/volk.c:27:
/Users/sabae/src/gnuradio/gnuradio/volk/build/lib/volk_machines.h:153:11: error: unknown type name
      'p_32f_8u_polarbutterfly_32f'
    const p_32f_8u_polarbutterfly_32f volk_32f_8u_polarbutterfly_32f_impls[22];
          ^
```

Here is [full output from `cmake` and `make`](https://gist.github.com/staticfloat/3015d151513919a844a4).  This happens on both the version of `volk` on the `maint` branch of gnuradio, as well as the `master` branch of `volk`.

n-west, 2015-10-08:
@michaelld have you seen this before? Some(?) function pointers aren't getting typedef'd and/or included properly. I'm not familiar with the OS X build process, but the compiler is AppleClang-- is that what you use?

@staticfloat do you see the missing typedefs in build/include/volk/volk_typedefs.h ?

michaelld, 2015-10-08:
@n-west : yes, I've seen something like that before. Racking my brain to remember what the issue was. In the meantime ...

@staticfloat : can you do the following & post the result. From the build directory:
```
rm -rf *
cmake ..
make VERBOSE=on
```
and, also post the file build/CMakeCache.txt .

staticfloat, 2015-10-08:
I'm building only `volk` right now, to simplify things, just so that's understood.

@n-west I do see it:
```
$ grep p_32f_8u_polarbutterfly_32f include/volk/volk_typedefs.h
typedef void (*p_32f_8u_polarbutterfly_32f)(float* , unsigned char* , const int , const int , const int , const int , const int );
```

Here's [the whole file](https://gist.github.com/staticfloat/282dab2e6c54ea5fdffe), for your perusal.

@michaelld: [Here are the outputs](https://gist.github.com/staticfloat/8d554a8804f1f6438ff9) you requested.

staticfloat, 2015-10-08:
I should note that using GCC 5.2.0 from Homebrew doesn't fix this, (done by setting `CC=gcc-5 CXX=g++-5` in my environment before running `cmake`)

michaelld, 2015-10-08:
@staticfloat : is Volk already installed somewhere, maybe /usr/local/include/volk ?

staticfloat, 2015-10-08:
Aha, yes, I have it in `~/local/include/volk` from a previous install.  Cleaning that out fixed this, thank you!

closed 2015-10-08 by staticfloat

michaelld, 2015-10-08:
NP. Glad that worked. As I like to say: cruft happens!