gnuradio/volk #52: volk_32f_cos_32f/volk_32f_sin_32f error

hutiantian, 2015-10-16:
output phase error
input is -38.1703:-0.0746:-53.0061
please see the result in the attachment 

[image]

hutiantian, 2015-10-16:
volk version is 1.1
and kernel is sse4.1 with alignment

drmpeg, 2015-10-18:
Seeing the same issue here in gr-paint. A random angle is created, and volk_32f_cos_32f and volk_32f_sin_32f are used to generate the constellation point.

[image]

Forcing volk_32f_cos_32f and volk_32f_sin_32f to generic in volk_config generates the correct constellation.

[image]

n-west, 2015-10-19:
Yikes! That looks pretty bad. Thanks for the report. I'll look in to this. Just to clarify, it's passing QA and volk_profile gives no errors, but the result is negative for those 3 chunks of angles?

drmpeg, 2015-10-19:
Here's the error in tabular format.

angle =   0, 0.000000, sin = 0.000000, volk sin = 0.000000, cos = 1.000000, volk cos = 1.000000
angle =   1, 0.017453, sin = 0.017452, volk sin = 0.017452, cos = 0.999848, volk cos = 0.999848
angle =   2, 0.034907, sin = 0.034899, volk sin = 0.034899, cos = 0.999391, volk cos = 0.999391
angle =   3, 0.052360, sin = 0.052336, volk sin = 0.052336, cos = 0.998630, volk cos = 0.998630
angle =   4, 0.069813, sin = 0.069756, volk sin = 0.069756, cos = 0.997564, volk cos = 0.997564
angle =   5, 0.087266, sin = 0.087156, volk sin = 0.087156, cos = 0.996195, volk cos = 0.996195
angle =   6, 0.104720, sin = 0.104528, volk sin = 0.104528, cos = 0.994522, volk cos = 0.994522
angle =   7, 0.122173, sin = 0.121869, volk sin = 0.121869, cos = 0.992546, volk cos = 0.992546
angle =   8, 0.139626, sin = 0.139173, volk sin = 0.139173, cos = 0.990268, volk cos = 0.990268
angle =   9, 0.157080, sin = 0.156434, volk sin = 0.156434, cos = 0.987688, volk cos = 0.987688
angle =  10, 0.174533, sin = 0.173648, volk sin = 0.173648, cos = 0.984808, volk cos = 0.984808
angle =  11, 0.191986, sin = 0.190809, volk sin = 0.190809, cos = 0.981627, volk cos = 0.981627
angle =  12, 0.209440, sin = 0.207912, volk sin = 0.207912, cos = 0.978148, volk cos = 0.978148
angle =  13, 0.226893, sin = 0.224951, volk sin = 0.224951, cos = 0.974370, volk cos = 0.974370
angle =  14, 0.244346, sin = 0.241922, volk sin = 0.241922, cos = 0.970296, volk cos = 0.970296
angle =  15, 0.261799, sin = 0.258819, volk sin = 0.258819, cos = 0.965926, volk cos = 0.965926
angle =  16, 0.279253, sin = 0.275637, volk sin = 0.275637, cos = 0.961262, volk cos = 0.961262
angle =  17, 0.296706, sin = 0.292372, volk sin = 0.292372, cos = 0.956305, volk cos = 0.956305
angle =  18, 0.314159, sin = 0.309017, volk sin = 0.309017, cos = 0.951057, volk cos = 0.951056
angle =  19, 0.331613, sin = 0.325568, volk sin = 0.325568, cos = 0.945519, volk cos = 0.945519
angle =  20, 0.349066, sin = 0.342020, volk sin = 0.342020, cos = 0.939693, volk cos = 0.939693
angle =  21, 0.366519, sin = 0.358368, volk sin = 0.358368, cos = 0.933580, volk cos = 0.933580
angle =  22, 0.383972, sin = 0.374607, volk sin = 0.374607, cos = 0.927184, volk cos = 0.927184
angle =  23, 0.401426, sin = 0.390731, volk sin = 0.390731, cos = 0.920505, volk cos = 0.920505
angle =  24, 0.418879, sin = 0.406737, volk sin = 0.406737, cos = 0.913545, volk cos = 0.913545
angle =  25, 0.436332, sin = 0.422618, volk sin = 0.422618, cos = 0.906308, volk cos = 0.906308
angle =  26, 0.453786, sin = 0.438371, volk sin = 0.438371, cos = 0.898794, volk cos = 0.898794
angle =  27, 0.471239, sin = 0.453991, volk sin = 0.453991, cos = 0.891007, volk cos = 0.891006
angle =  28, 0.488692, sin = 0.469472, volk sin = 0.469472, cos = 0.882948, volk cos = 0.882948
angle =  29, 0.506145, sin = 0.484810, volk sin = 0.484810, cos = 0.874620, volk cos = 0.874620
angle =  30, 0.523599, sin = 0.500000, volk sin = 0.500000, cos = 0.866025, volk cos = 0.866025
angle =  31, 0.541052, sin = 0.515038, volk sin = 0.515038, cos = 0.857167, volk cos = 0.857167
angle =  32, 0.558505, sin = 0.529919, volk sin = 0.529919, cos = 0.848048, volk cos = 0.848048
angle =  33, 0.575959, sin = 0.544639, volk sin = 0.544639, cos = 0.838671, volk cos = 0.838671
angle =  34, 0.593412, sin = 0.559193, volk sin = 0.559193, cos = 0.829038, volk cos = 0.829037
angle =  35, 0.610865, sin = 0.573576, volk sin = 0.573577, cos = 0.819152, volk cos = 0.819152
angle =  36, 0.628319, sin = 0.587785, volk sin = 0.587785, cos = 0.809017, volk cos = 0.809017
angle =  37, 0.645772, sin = 0.601815, volk sin = 0.601815, cos = 0.798636, volk cos = 0.798635
angle =  38, 0.663225, sin = 0.615661, volk sin = 0.615662, cos = 0.788011, volk cos = 0.788011
angle =  39, 0.680678, sin = 0.629320, volk sin = 0.629321, cos = 0.777146, volk cos = 0.777146
angle =  40, 0.698132, sin = 0.642788, volk sin = 0.642788, cos = 0.766044, volk cos = 0.766044
angle =  41, 0.715585, sin = 0.656059, volk sin = 0.656059, cos = 0.754710, volk cos = 0.754710
angle =  42, 0.733038, sin = 0.669131, volk sin = 0.669131, cos = 0.743145, volk cos = 0.743145
angle =  43, 0.750492, sin = 0.681998, volk sin = 0.681998, cos = 0.731354, volk cos = 0.731354
angle =  44, 0.767945, sin = 0.694658, volk sin = 0.694658, cos = 0.719340, volk cos = 0.719340
angle =  45, 0.785398, sin = 0.707107, volk sin = 0.707107, cos = 0.707107, volk cos = 0.707107
angle =  46, 0.802851, sin = 0.719340, volk sin = 0.719340, cos = 0.694658, volk cos = 0.694658
angle =  47, 0.820305, sin = 0.731354, volk sin = 0.731354, cos = 0.681998, volk cos = 0.681998
angle =  48, 0.837758, sin = 0.743145, volk sin = 0.743145, cos = 0.669131, volk cos = 0.669131
angle =  49, 0.855211, sin = 0.754710, volk sin = 0.754710, cos = 0.656059, volk cos = 0.656059
angle =  50, 0.872665, sin = 0.766044, volk sin = 0.766044, cos = 0.642788, volk cos = 0.642788
angle =  51, 0.890118, sin = 0.777146, volk sin = 0.777146, cos = 0.629320, volk cos = 0.629320
angle =  52, 0.907571, sin = 0.788011, volk sin = 0.788011, cos = 0.615661, volk cos = 0.615661
angle =  53, 0.925025, sin = 0.798636, volk sin = 0.798636, cos = 0.601815, volk cos = 0.601815
angle =  54, 0.942478, sin = 0.809017, volk sin = 0.809017, cos = 0.587785, volk cos = 0.587785
angle =  55, 0.959931, sin = 0.819152, volk sin = 0.819152, cos = 0.573576, volk cos = 0.573576
[... cut ...]
angle = 355, 6.195919, sin = -0.087156, volk sin = 0.087155, cos = 0.996195, volk cos = 0.996195 "
angle = 356, 6.213372, sin = -0.069756, volk sin = 0.069756, cos = 0.997564, volk cos = 0.997564 "
angle = 357, 6.230825, sin = -0.052336, volk sin = 0.052336, cos = 0.998630, volk cos = 0.998630 "
angle = 358, 6.248279, sin = -0.034900, volk sin = 0.034899, cos = 0.999391, volk cos = 0.999391 "
angle = 359, 6.265732, sin = -0.017453, volk sin = 0.017452, cos = 0.999848, volk cos = 0.999848 sine error*

n-west, 2015-11-04:
OK, just got around to this today. My first step whenever I look at this stuff is to figure out why our QA fails. In this case it's because our auto-generated numbers are (-1,1). When I change that to (-pi, pi) I get failures for the trig functions (tan is also off, tanh is too, but that's just a tolerance problem for high input vals).

Unfortunately I haven't seen an obvious transition/pattern with where the sign errors are (please tell me if you do, which may help exploring the math). I didn't actually write the original code, so it'll take another night of re-deriving the appromixations.

drmpeg, 2015-11-04:
For cosine, the sign errors are from 3/8 pi to 1/2 pi and 11/8 pi to 3/2 pi.

For sine, the sign errors are from 7/8 pi to pi and 15/8 pi to 2 pi.

stwunsch, 2015-11-29:
Just looked through the code and tried to fix it, no luck :/ But I don't think that sse 4.1 is necessary, sse 2 should be sufficient for this kernel. I could be wrong!

n-west, 2015-12-01:
@stwunsch, I *think* you're correct that we only need SSE2 for this. I'd need to look at operations another time. For now, I found the paper this is based on and discovered a missing floor operation on a value used in a if-branch to flip the sign. This should be fixed. Sorry for the long delay, and I appreciate the patience :-)

While I was at it I updated the QA to test from [-pi, pi] so we test all around the unit circle. As a result these moved in to the "inaccurate" section of kernels because the normal tolerance is quite tight, but in reality these are very good approximations.

I'm going to close this since I think the issue is fixed. Holler if you see any issues.

closed 2015-12-01 by n-west

n-west, 2015-12-01:
Heh, ok... This will still fail QA because the randomly generated numbers provide some really small angles and the % difference used in the QA makes it tough to do these comparisons. That's been on my TODO list for a while and it looks like I have a good reason to do it now..

For now ignore really small errors even though the QA will fail. Example:
1: offset 10269 in1: -7.58967e-07 in2: -8.34465e-07 tolerance was: 0.001. The difference is more decimal places out than we normally care, but since we use % difference it's a rather large error.

stwunsch, 2015-12-01:
Can you post the name of the paper? Thx in advance!

n-west, 2015-12-01:
Efficient evaluation methods of elementary functions suitable for SIMD computation.
Link: http://link.springer.com/article/10.1007%2Fs00450-010-0108-2

The basic idea is to get the form s + pi/4 * q, where q is an integer that represents the quadrant, then put use cosine double angle identity (3 times in this case) to put s in to a smaller range so it's good for a taylor series approximation and use q to recover the correct sign. Taking advantage of the double-angle multiple times (along with the math to unwind) to get a nicely linearized region is the clever bit.