gnuradio/volk #53: test_all fails for volk_32fc_magnitude_32f and volk_32f_sqrt_32f on neon

djanderson, 2015-11-12:
```bash
$ uname -a
Linux odroid 3.10.82-57 #1 SMP PREEMPT Wed Oct 7 13.19:01 BRT 2015 armv7l armv7l armv7l GNU/Linux
```
```
Start testing: Nov 12 18:22 UTC
----------------------------------------------------------
1/1 Testing: qa_test_all
1/1 Test: qa_test_all
Command: "/bin/sh" "/home/odroid/Desktop/volk/build/lib/test_all_test.sh" "/home/odroid/Desktop/volk/build/lib" "/home/odroid/Desktop/volk/build/lib"
Directory: /home/odroid/Desktop/volk/build/lib
"qa_test_all" start time: Nov 12 18:22 UTC
Output:
----------------------------------------------------------
Using Volk machine: neon_hardfp_orc
RUN_VOLK_TESTS: volk_64u_popcntpuppet_64u(131071,1)
generic completed in 1.36ms
neon completed in 1.572ms
Best aligned arch: generic
Best unaligned arch: generic
RUN_VOLK_TESTS: volk_16u_byteswappuppet_16u(131071,1)
generic completed in 0.376ms
neon completed in 0.338ms
neon_table completed in 0.361ms
Best aligned arch: neon
Best unaligned arch: neon
RUN_VOLK_TESTS: volk_32u_byteswappuppet_32u(131071,1)
generic completed in 0.588ms
neon completed in 0.558ms
Best aligned arch: neon
Best unaligned arch: neon
RUN_VOLK_TESTS: volk_32u_popcntpuppet_32u(131071,1)
generic completed in 0.862ms
Best aligned arch: generic
Best unaligned arch: generic
RUN_VOLK_TESTS: volk_64u_byteswappuppet_64u(131071,1)
generic completed in 1.413ms
neon completed in 1.201ms
Best aligned arch: neon
Best unaligned arch: neon
RUN_VOLK_TESTS: volk_32fc_s32fc_rotatorpuppet_32fc(131071,1)
generic completed in 9.563ms
Best aligned arch: generic
Best unaligned arch: generic
RUN_VOLK_TESTS: volk_8u_conv_k7_r2puppet_8u(131071,0)
generic completed in 0.004ms
Best aligned arch: generic
Best unaligned arch: generic
RUN_VOLK_TESTS: volk_32f_x2_fm_detectpuppet_32f(131071,1)
generic completed in 2.002ms
Best aligned arch: generic
Best unaligned arch: generic
RUN_VOLK_TESTS: volk_16ic_s32f_deinterleave_real_32f(131071,1)
generic completed in 0.773ms
Best aligned arch: generic
Best unaligned arch: generic
RUN_VOLK_TESTS: volk_16ic_deinterleave_real_8i(131071,1)
generic completed in 0.42ms
neon completed in 0.407ms
u_orc completed in 2.634ms
Best aligned arch: neon
Best unaligned arch: neon
RUN_VOLK_TESTS: volk_16ic_deinterleave_16i_x2(131071,1)
generic completed in 0.638ms
u_orc completed in 0.911ms
Best aligned arch: generic
Best unaligned arch: generic
RUN_VOLK_TESTS: volk_16ic_s32f_deinterleave_32f_x2(131071,1)
generic completed in 0.932ms
neon completed in 0.928ms
[... 444 lines cut ...]
series completed in 1.362ms
Best aligned arch: series
Best unaligned arch: series
RUN_VOLK_TESTS: volk_8u_x3_encodepolarpuppet_8u(131071,1)
generic completed in 1.611ms
Best aligned arch: generic
Best unaligned arch: generic
RUN_VOLK_TESTS: volk_32f_8u_polarbutterflypuppet_32f(131071,1)
generic completed in 9.503ms
Best aligned arch: generic
Best unaligned arch: generic
Kernel QA finished: 2 failures out of 93 tests.
The following kernels failed QA:
    volk_32fc_magnitude_32f
    volk_32f_sqrt_32f
<end of output>
Test time =   6.35 sec
----------------------------------------------------------
Test Failed.
"qa_test_all" end time: Nov 12 18:22 UTC
"qa_test_all" time elapsed: 00:00:06
----------------------------------------------------------

End testing: Nov 12 18:22 UTC
```

douggeiger, 2015-11-19:
This looks like a tolerances issue: i.e. all of the errors are within a few decimal points - so some of the NEON implementations are not with whatever tolerance was specified in the QA checks.

n-west, 2015-11-20:
This is just a tolerance issue. Here is what I've said about it in the past:

"The other two have nothing 'wrong' with them. We use a QA tolerance to check results and it just happens that the tolerance is a bit too low on ARM. Those operations are hard to get accurate and fast because they involve a square root. NEON happens to have an inverse square root estimate which we use because it's the fastest way to get a result. Unfortunately whenever we have to use the inverse square root estimate from NEON the accuracy drops a bit. In these cases since we actually care about the sqiare root we use the NEON inverse estimate which is also not particularly accurate. The two together make any square root-like operations in VOLK not particularly accurate in NEON. On the plus side it's probably accurate enough for radio."

I pushed an update tolerance value for QA to these two kernels to my personal copy and will likely merge it in to maint tonight or (more likely) tomorrow.

Thanks for the bug report and taking time to run QA on embedded platforms!

closed 2015-12-23 by n-west