gnuradio/volk #55: gnss-sdr build fails with volk-1.1.1

wkazubski, 2015-11-20:
Building gnss-sdr-0.0.6 (https://github.com/gnss-sdr/gnss-sdr) with GNU Radio-3.7.8.1 (libvolk-1.1.1) fails with a number of warnings. With previous version of GNU Radio (libvolk-1.0.2) everything was OK. 

The warnigs are:
<code>
...
Building CXX object src/algorithms/tracking/gnuradio_blocks/CMakeFiles/tracking_gr_blocks.dir/galileo_e1_tcp_connector_tracking_cc.cc.o     
In file included from /home/wk/rpmbuild/BUILD/gnss-sdr-0.0.6/build/volk_gnsssdr_module/build/include/volk_gnsssdr/volk_gnsssdr.h:27:0,
                 from /home/wk/rpmbuild/BUILD/gnss-sdr-0.0.6/src/algorithms/tracking/gnuradio_blocks/galileo_volk_e1_dll_pll_veml_tracking_cc.cc:51:
/home/wk/rpmbuild/BUILD/gnss-sdr-0.0.6/build/volk_gnsssdr_module/build/include/volk_gnsssdr/volk_gnsssdr_config_fixed.h:28:0: warning: "LV_SOFTFP" redefined [enabled by default]
 #define LV_SOFTFP 2
 ^
In file included from /usr/include/volk/volk.h:30:0,
                 from /home/wk/rpmbuild/BUILD/gnss-sdr-0.0.6/src/algorithms/tracking/libs/correlator.h:41,
                 from /home/wk/rpmbuild/BUILD/gnss-sdr-0.0.6/src/algorithms/tracking/gnuradio_blocks/galileo_volk_e1_dll_pll_veml_tracking_cc.h:47,
                 from /home/wk/rpmbuild/BUILD/gnss-sdr-0.0.6/src/algorithms/tracking/gnuradio_blocks/galileo_volk_e1_dll_pll_veml_tracking_cc.cc:37:
/usr/include/volk/volk_config_fixed.h:30:0: note: this is the location of the previous definition
 #define LV_SOFTFP 1
 ^
In file included from /home/wk/rpmbuild/BUILD/gnss-sdr-0.0.6/build/volk_gnsssdr_module/build/include/volk_gnsssdr/volk_gnsssdr.h:27:0,
                 from /home/wk/rpmbuild/BUILD/gnss-sdr-0.0.6/src/algorithms/tracking/gnuradio_blocks/galileo_volk_e1_dll_pll_veml_tracking_cc.cc:51:
/home/wk/rpmbuild/BUILD/gnss-sdr-0.0.6/build/volk_gnsssdr_module/build/include/volk_gnsssdr/volk_gnsssdr_config_fixed.h:29:0: warning: "LV_HARDFP" redefined [enabled by default]
 #define LV_HARDFP 3
 ^
In file included from /usr/include/volk/volk.h:30:0,
                 from /home/wk/rpmbuild/BUILD/gnss-sdr-0.0.6/src/algorithms/tracking/libs/correlator.h:41,
                 from /home/wk/rpmbuild/BUILD/gnss-sdr-0.0.6/src/algorithms/tracking/gnuradio_blocks/galileo_volk_e1_dll_pll_veml_tracking_cc.h:47,
                 from /home/wk/rpmbuild/BUILD/gnss-sdr-0.0.6/src/algorithms/tracking/gnuradio_blocks/galileo_volk_e1_dll_pll_veml_tracking_cc.cc:37:
/usr/include/volk/volk_config_fixed.h:31:0: note: this is the location of the previous definition
 #define LV_HARDFP 2
 ^
In file included from /home/wk/rpmbuild/BUILD/gnss-sdr-0.0.6/build/volk_gnsssdr_module/build/include/volk_gnsssdr/volk_gnsssdr.h:27:0,
                 from /home/wk/rpmbuild/BUILD/gnss-sdr-0.0.6/src/algorithms/tracking/gnuradio_blocks/galileo_volk_e1_dll_pll_veml_tracking_cc.cc:51:
/home/wk/rpmbuild/BUILD/gnss-sdr-0.0.6/build/volk_gnsssdr_module/build/include/volk_gnsssdr/volk_gnsssdr_config_fixed.h:30:0: warning: "LV_NEON" redefined [enabled by default]
 #define LV_NEON 4
 ^
In file included from /usr/include/volk/volk.h:30:0,
                 from /home/wk/rpmbuild/BUILD/gnss-sdr-0.0.6/src/algorithms/tracking/libs/correlator.h:41,
                 from /home/wk/rpmbuild/BUILD/gnss-sdr-0.0.6/src/algorithms/tracking/gnuradio_blocks/galileo_volk_e1_dll_pll_veml_tracking_cc.h:47,
                 from /home/wk/rpmbuild/BUILD/gnss-sdr-0.0.6/src/algorithms/tracking/gnuradio_blocks/galileo_volk_e1_dll_pll_veml_tracking_cc.cc:37:
/usr/include/volk/volk_config_fixed.h:32:0: note: this is the location of the previous definition
 #define LV_NEON 3
 ^
In file included from /home/wk/rpmbuild/BUILD/gnss-sdr-0.0.6/build/volk_gnsssdr_module/build/include/volk_gnsssdr/volk_gnsssdr.h:27:0,
                 from /home/wk/rpmbuild/BUILD/gnss-sdr-0.0.6/src/algorithms/tracking/gnuradio_blocks/galileo_volk_e1_dll_pll_veml_tracking_cc.cc:51:
/home/wk/rpmbuild/BUILD/gnss-sdr-0.0.6/build/volk_gnsssdr_module/build/include/volk_gnsssdr/volk_gnsssdr_config_fixed.h:31:0: warning: "LV_32" redefined [enabled by default]
 #define LV_32 5
 ^
In file included from /usr/include/volk/volk.h:30:0,
                 from /home/wk/rpmbuild/BUILD/gnss-sdr-0.0.6/src/algorithms/tracking/libs/correlator.h:41,
                 from /home/wk/rpmbuild/BUILD/gnss-sdr-0.0.6/src/algorithms/tracking/gnuradio_blocks/galileo_volk_e1_dll_pll_veml_tracking_cc.h:47,
                 from /home/wk/rpmbuild/BUILD/gnss-sdr-0.0.6/src/algorithms/tracking/gnuradio_blocks/galileo_volk_e1_dll_pll_veml_tracking_cc.cc:37:
/usr/include/volk/volk_config_fixed.h:33:0: note: this is the location of the previous definition
 #define LV_32 4
 ^
In file included from /home/wk/rpmbuild/BUILD/gnss-sdr-0.0.6/build/volk_gnsssdr_module/build/include/volk_gnsssdr/volk_gnsssdr.h:27:0,
                 from /home/wk/rpmbuild/BUILD/gnss-sdr-0.0.6/src/algorithms/tracking/gnuradio_blocks/galileo_volk_e1_dll_pll_veml_tracking_cc.cc:51:
/home/wk/rpmbuild/BUILD/gnss-sdr-0.0.6/build/volk_gnsssdr_module/build/include/volk_gnsssdr/volk_gnsssdr_config_fixed.h:32:0: warning: "LV_64" redefined [enabled by default]
 #define LV_64 6
 ^
In file included from /usr/include/volk/volk.h:30:0,
                 from /home/wk/rpmbuild/BUILD/gnss-sdr-0.0.6/src/algorithms/tracking/libs/correlator.h:41,
                 from /home/wk/rpmbuild/BUILD/gnss-sdr-0.0.6/src/algorithms/tracking/gnuradio_blocks/galileo_volk_e1_dll_pll_veml_tracking_cc.h:47,
                 from /home/wk/rpmbuild/BUILD/gnss-sdr-0.0.6/src/algorithms/tracking/gnuradio_blocks/galileo_volk_e1_dll_pll_veml_tracking_cc.cc:37:
/usr/include/volk/volk_config_fixed.h:34:0: note: this is the location of the previous definition
 #define LV_64 5
 ^
In file included from /home/wk/rpmbuild/BUILD/gnss-sdr-0.0.6/build/volk_gnsssdr_module/build/include/volk_gnsssdr/volk_gnsssdr.h:27:0,
                 from /home/wk/rpmbuild/BUILD/gnss-sdr-0.0.6/src/algorithms/tracking/gnuradio_blocks/galileo_volk_e1_dll_pll_veml_tracking_cc.cc:51:
/home/wk/rpmbuild/BUILD/gnss-sdr-0.0.6/build/volk_gnsssdr_module/build/include/volk_gnsssdr/volk_gnsssdr_config_fixed.h:33:0: warning: "LV_3DNOW" redefined [enabled by default]
 #define LV_3DNOW 7
[... 77 lines cut ...]
                                                     ^
In file included from /usr/include/volk/volk_typedefs.h:30:0,
                 from /usr/include/volk/volk.h:29,
                 from /home/wk/rpmbuild/BUILD/gnss-sdr-0.0.6/src/algorithms/tracking/libs/correlator.h:41,
                 from /home/wk/rpmbuild/BUILD/gnss-sdr-0.0.6/src/algorithms/tracking/gnuradio_blocks/galileo_volk_e1_dll_pll_veml_tracking_cc.h:47,
                 from /home/wk/rpmbuild/BUILD/gnss-sdr-0.0.6/src/algorithms/tracking/gnuradio_blocks/galileo_volk_e1_dll_pll_veml_tracking_cc.cc:37:
/usr/include/volk/volk_complex.h:40:53: error: 'template<class T> typename T::value_type lv_cimag(const T&)' previously declared here
 template <typename T> inline typename T::value_type lv_cimag(const T &x){
                                                     ^
In file included from /home/wk/rpmbuild/BUILD/gnss-sdr-0.0.6/build/volk_gnsssdr_module/build/include/volk_gnsssdr/volk_gnsssdr_typedefs.h:27:0,
                 from /home/wk/rpmbuild/BUILD/gnss-sdr-0.0.6/build/volk_gnsssdr_module/build/include/volk_gnsssdr/volk_gnsssdr.h:26,
                 from /home/wk/rpmbuild/BUILD/gnss-sdr-0.0.6/src/algorithms/tracking/gnuradio_blocks/galileo_volk_e1_dll_pll_veml_tracking_cc.cc:51:
/home/wk/rpmbuild/BUILD/gnss-sdr-0.0.6/src/algorithms/libs/volk_gnsssdr_module/volk_gnsssdr/include/volk_gnsssdr/volk_gnsssdr_complex.h:67:32: error: redefinition of 'template<class T> T lv_conj(const T&)'
 template <typename T> inline T lv_conj(const T &x){
                                ^
In file included from /usr/include/volk/volk_typedefs.h:30:0,
                 from /usr/include/volk/volk.h:29,
                 from /home/wk/rpmbuild/BUILD/gnss-sdr-0.0.6/src/algorithms/tracking/libs/correlator.h:41,
                 from /home/wk/rpmbuild/BUILD/gnss-sdr-0.0.6/src/algorithms/tracking/gnuradio_blocks/galileo_volk_e1_dll_pll_veml_tracking_cc.h:47,
                 from /home/wk/rpmbuild/BUILD/gnss-sdr-0.0.6/src/algorithms/tracking/gnuradio_blocks/galileo_volk_e1_dll_pll_veml_tracking_cc.cc:37:
/usr/include/volk/volk_complex.h:44:32: error: 'template<class T> T lv_conj(const T&)' previously declared here
 template <typename T> inline T lv_conj(const T &x){
                                ^
...
</code>

wkazubski, 2015-11-20:
The warnings in a file:
[warnings.txt](https://github.com/gnuradio/volk/files/40218/warnings.txt)

n-west, 2015-11-20:
VOLK recently changed the ifdef guards in volk_complex. The side affect for previously existing VOLK modules is some errors on redefinitions. It looks like Carles fixed this in gnss-sdr (https://github.com/gnss-sdr/gnss-sdr/blob/master/src/algorithms/libs/volk_gnsssdr_module/volk_gnsssdr/include/volk_gnsssdr/volk_gnsssdr_complex.h#L27) in September. Can you confirm that you're using a version of gnss-sdr that includes that fix?

wkazubski, 2015-11-20:
After updating to latest master branch, I still see the warnings but the build continues.
Thanks for help.

closed 2015-12-23 by n-west