gnuradio/volk #59: Kernel name with leading x character throws exception

stwunsch, 2015-12-20:
Hi,

A kernel with the name volk_32f_x2_x<sth>_32f throws an exception during ctest. The problem is simply a leading x in the kernel name "x<sth>", which looks like the "x<number>" of the input signiture.
If this is known and intended, just close the issue!

Greetings
Stefan

stwunsch, 2015-12-20:
I mean this exception in qa_utils.cc:376:

    catch (boost::bad_lexical_cast& error) {
        std::cerr << "Error: unable to get function signature from kernel name" << std::endl;
        std::cerr << "  - " << name << std::endl;
        return false;
    }

n-west, 2015-12-20:
Hey Stefan, not intended but not something anyone has complained about. If the fix is as easy as I think it might be I can fix it, otherwise it's kind of a niche feature (especially the _x_ case) that it might be a do not fix.

douggeiger, 2015-12-20:
I'm guessing this is the signature parsing code not knowing when it's parsing signatures (ala _x2_ or _x3_ to signify multiple inputs of the same type) vs. when it is getting to the name of the kernel. I'm not confident there is an unamiguous way for the parser to know this in advance - except to check if the second character is a digit or not - which basically just makes x[0-9]+ a special (not-allowed at the beginning of the name) vs. just x* being special.

n-west, 2015-12-21:
Fixed

closed 2015-12-21 by n-west

ValZapod, 2020-01-13:
`(token[1] > '0' || token[1] < '9')`
@michaelld There should be && in this commit.

michaelld, 2020-01-13:
@ValZapod Let's take the discussion to this issue https://github.com/gnuradio/volk/issues/326

ValZapod, 2020-01-13:
@michaelld https://github.com/gnuradio/volk/issues/59#issuecomment-166156861 Here it is said x[0-9]