gnuradio/volk #60: 32f_{sin,cos,tan}_32f Fail on

mcq, 2016-01-07:
git clone (to master at c2c7f82ae)
cmake .
make
cd lib
test_all
(on an AMD A4-5000)

Kernel QA finished: 3 failures out of 93 tests.
The following kernels failed QA:
    volk_32f_sin_32f
    volk_32f_cos_32f
    volk_32f_tan_32f

---earlier:
offset 21089 in1: -0.000102909 in2: -0.00010276 tolerance was: 0.001
offset 23287 in1: 2.16087e-05 in2: 2.14591e-05 tolerance was: 0.001
offset 81567 in1: 0.000114115 in2: 0.000113965 tolerance was: 0.001
volk_32f_sin_32f: fail on arch a_sse4_1
offset 21089 in1: -0.000102909 in2: -0.00010276 tolerance was: 0.001
offset 23287 in1: 2.16087e-05 in2: 2.14591e-05 tolerance was: 0.001
offset 81567 in1: 0.000114115 in2: 0.000113965 tolerance was: 0.001
volk_32f_sin_32f: fail on arch u_sse4_1
...
offset 10269 in1: -7.58967e-07 in2: -8.34465e-07 tolerance was: 0.001
offset 15690 in1: 5.09779e-05 in2: 5.09024e-05 tolerance was: 0.001
offset 101940 in1: -3.4972e-05 in2: -3.50475e-05 tolerance was: 0.001
offset 114392 in1: 4.72466e-06 in2: 4.64916e-06 tolerance was: 0.001
offset 130159 in1: 4.64479e-05 in2: 4.63724e-05 tolerance was: 0.001
volk_32f_cos_32f: fail on arch a_sse4_1
offset 10269 in1: -7.58967e-07 in2: -8.34465e-07 tolerance was: 0.001
offset 15690 in1: 5.09779e-05 in2: 5.09024e-05 tolerance was: 0.001
offset 101940 in1: -3.4972e-05 in2: -3.50475e-05 tolerance was: 0.001
offset 114392 in1: 4.72466e-06 in2: 4.64916e-06 tolerance was: 0.001
offset 130159 in1: 4.64479e-05 in2: 4.63724e-05 tolerance was: 0.001
volk_32f_cos_32f: fail on arch u_sse4_1
offset 10269 in1: -7.58967e-07 in2: -8.18572e-07 tolerance was: 0.001
offset 15690 in1: 5.09779e-05 in2: 5.09183e-05 tolerance was: 0.001
offset 101940 in1: -3.4972e-05 in2: -3.50316e-05 tolerance was: 0.001
offset 114392 in1: 4.72466e-06 in2: 4.66506e-06 tolerance was: 0.001
offset 130159 in1: 4.64479e-05 in2: 4.63883e-05 tolerance was: 0.001
volk_32f_cos_32f: fail on arch generic_fast
...
offset 8630 in1: -3.32912e-05 in2: -3.31416e-05 tolerance was: 0.001
offset 14489 in1: 15898.6 in2: 15917.7 tolerance was: 0.001
offset 28068 in1: 9.8141e-05 in2: 9.79914e-05 tolerance was: 0.001
offset 31758 in1: -0.000106247 in2: -0.000106098 tolerance was: 0.001
offset 44343 in1: 189075 in2: 186414 tolerance was: 0.001
offset 93063 in1: -6.97692e-05 in2: -6.96196e-05 tolerance was: 0.001
volk_32f_tan_32f: fail on arch a_sse4_1
offset 8630 in1: -3.32912e-05 in2: -3.31416e-05 tolerance was: 0.001
offset 14489 in1: 15898.6 in2: 15917.7 tolerance was: 0.001
offset 28068 in1: 9.8141e-05 in2: 9.79914e-05 tolerance was: 0.001
offset 31758 in1: -0.000106247 in2: -0.000106098 tolerance was: 0.001
offset 44343 in1: 189075 in2: 186414 tolerance was: 0.001
offset 93063 in1: -6.97692e-05 in2: -6.96196e-05 tolerance was: 0.001
volk_32f_tan_32f: fail on arch u_sse4_1
...
offset 21089 in1: -0.000102909 in2: -0.00010276 tolerance was: 0.001
offset 23287 in1: 2.16087e-05 in2: 2.14591e-05 tolerance was: 0.001  
offset 81567 in1: 0.000114115 in2: 0.000113965 tolerance was: 0.001
volk_32f_sin_32f: fail on arch a_sse4_1
offset 21089 in1: -0.000102909 in2: -0.00010276 tolerance was: 0.001
offset 23287 in1: 2.16087e-05 in2: 2.14591e-05 tolerance was: 0.001
offset 81567 in1: 0.000114115 in2: 0.000113965 tolerance was: 0.001  
volk_32f_sin_32f: fail on arch u_sse4_1
...
offset 10269 in1: -7.58967e-07 in2: -8.34465e-07 tolerance was: 0.001
offset 15690 in1: 5.09779e-05 in2: 5.09024e-05 tolerance was: 0.001
offset 101940 in1: -3.4972e-05 in2: -3.50475e-05 tolerance was: 0.001
offset 114392 in1: 4.72466e-06 in2: 4.64916e-06 tolerance was: 0.001
offset 130159 in1: 4.64479e-05 in2: 4.63724e-05 tolerance was: 0.001
volk_32f_cos_32f: fail on arch a_sse4_1
offset 10269 in1: -7.58967e-07 in2: -8.34465e-07 tolerance was: 0.001
offset 15690 in1: 5.09779e-05 in2: 5.09024e-05 tolerance was: 0.001
offset 101940 in1: -3.4972e-05 in2: -3.50475e-05 tolerance was: 0.001
offset 114392 in1: 4.72466e-06 in2: 4.64916e-06 tolerance was: 0.001
offset 130159 in1: 4.64479e-05 in2: 4.63724e-05 tolerance was: 0.001
volk_32f_cos_32f: fail on arch u_sse4_1
offset 10269 in1: -7.58967e-07 in2: -8.18572e-07 tolerance was: 0.001
offset 15690 in1: 5.09779e-05 in2: 5.09183e-05 tolerance was: 0.001
offset 101940 in1: -3.4972e-05 in2: -3.50316e-05 tolerance was: 0.001
offset 114392 in1: 4.72466e-06 in2: 4.66506e-06 tolerance was: 0.001
offset 130159 in1: 4.64479e-05 in2: 4.63883e-05 tolerance was: 0.001
volk_32f_cos_32f: fail on arch generic_fast
...
offset 8630 in1: -3.32912e-05 in2: -3.31416e-05 tolerance was: 0.001
offset 14489 in1: 15898.6 in2: 15917.7 tolerance was: 0.001
offset 28068 in1: 9.8141e-05 in2: 9.79914e-05 tolerance was: 0.001
offset 31758 in1: -0.000106247 in2: -0.000106098 tolerance was: 0.001
offset 44343 in1: 189075 in2: 186414 tolerance was: 0.001
offset 93063 in1: -6.97692e-05 in2: -6.96196e-05 tolerance was: 0.001
volk_32f_tan_32f: fail on arch a_sse4_1
offset 8630 in1: -3.32912e-05 in2: -3.31416e-05 tolerance was: 0.001
offset 14489 in1: 15898.6 in2: 15917.7 tolerance was: 0.001
offset 28068 in1: 9.8141e-05 in2: 9.79914e-05 tolerance was: 0.001
offset 31758 in1: -0.000106247 in2: -0.000106098 tolerance was: 0.001
offset 44343 in1: 189075 in2: 186414 tolerance was: 0.001
offset 93063 in1: -6.97692e-05 in2: -6.96196e-05 tolerance was: 0.001
volk_32f_tan_32f: fail on arch u_sse4_1

davidjmyers, 2016-01-12:
I'm running into similar issues.
I'm also a little confused by my ctest output because it seems to indicate that Volk is selecting the failed alignments as the "Best aligned arch[s]".

Here's my set up:
Intel® Core™ i7-5600U CPU @ 2.60GHz × 4
Ubuntu 14.04 LTS
GNURadio (v3.7.9) @ git commit dec480ab3f0809677ba3ef2a3a64d402d742b5ec
Volk @ git tag: v1.2

Output of ctest -V (after running make on GNU Radio):

1: RUN_VOLK_TESTS: volk_32f_sin_32f(131071,1)
1: a_sse4_1 completed in 0.548ms
1: u_sse4_1 completed in 0.572ms
1: generic completed in 3.516ms
1: offset 21089 in1: -0.000102909 in2: -0.00010276 tolerance was: 0.001
1: offset 23287 in1: 2.16087e-05 in2: 2.14591e-05 tolerance was: 0.001
1: offset 81567 in1: 0.000114115 in2: 0.000113965 tolerance was: 0.001
1: volk_32f_sin_32f: fail on arch a_sse4_1
1: offset 21089 in1: -0.000102909 in2: -0.00010276 tolerance was: 0.001
1: offset 23287 in1: 2.16087e-05 in2: 2.14591e-05 tolerance was: 0.001
1: offset 81567 in1: 0.000114115 in2: 0.000113965 tolerance was: 0.001
1: volk_32f_sin_32f: fail on arch u_sse4_1
1: Best aligned arch: a_sse4_1
1: Best unaligned arch: u_sse4_1
1: Failure on volk_32f_sin_32f
1: RUN_VOLK_TESTS: volk_32f_cos_32f(131071,1)
1: a_sse4_1 completed in 0.542ms
1: u_sse4_1 completed in 0.619ms
1: generic_fast completed in 2.962ms
1: generic completed in 4.38ms
1: offset 10269 in1: -7.58967e-07 in2: -8.34465e-07 tolerance was: 0.001
1: offset 15690 in1: 5.09779e-05 in2: 5.09024e-05 tolerance was: 0.001
1: offset 101940 in1: -3.4972e-05 in2: -3.50475e-05 tolerance was: 0.001
1: offset 114392 in1: 4.72466e-06 in2: 4.64916e-06 tolerance was: 0.001
1: offset 130159 in1: 4.64479e-05 in2: 4.63724e-05 tolerance was: 0.001
1: volk_32f_cos_32f: fail on arch a_sse4_1
1: offset 10269 in1: -7.58967e-07 in2: -8.34465e-07 tolerance was: 0.001
1: offset 15690 in1: 5.09779e-05 in2: 5.09024e-05 tolerance was: 0.001
1: offset 101940 in1: -3.4972e-05 in2: -3.50475e-05 tolerance was: 0.001
1: offset 114392 in1: 4.72466e-06 in2: 4.64916e-06 tolerance was: 0.001
1: offset 130159 in1: 4.64479e-05 in2: 4.63724e-05 tolerance was: 0.001
1: volk_32f_cos_32f: fail on arch u_sse4_1
1: offset 10269 in1: -7.58967e-07 in2: -8.18572e-07 tolerance was: 0.001
1: offset 15690 in1: 5.09779e-05 in2: 5.09183e-05 tolerance was: 0.001
1: offset 101940 in1: -3.4972e-05 in2: -3.50316e-05 tolerance was: 0.001
1: offset 114392 in1: 4.72466e-06 in2: 4.66506e-06 tolerance was: 0.001
1: offset 130159 in1: 4.64479e-05 in2: 4.63883e-05 tolerance was: 0.001
1: volk_32f_cos_32f: fail on arch generic_fast
1: Best aligned arch: a_sse4_1
1: Best unaligned arch: u_sse4_1
1: RUN_VOLK_TESTS: volk_32f_tan_32f(131071,1)
1: Failure on volk_32f_cos_32f
1: a_sse4_1 completed in 0.709ms
1: u_sse4_1 completed in 0.771ms
1: generic completed in 5.333ms
1: offset 8630 in1: -3.32912e-05 in2: -3.31416e-05 tolerance was: 0.001
1: offset 14489 in1: 15898.6 in2: 15917.7 tolerance was: 0.001
1: offset 28068 in1: 9.8141e-05 in2: 9.79914e-05 tolerance was: 0.001
1: offset 31758 in1: -0.000106247 in2: -0.000106098 tolerance was: 0.001
1: offset 44343 in1: 189075 in2: 186414 tolerance was: 0.001
1: offset 93063 in1: -6.97692e-05 in2: -6.96196e-05 tolerance was: 0.001
1: volk_32f_tan_32f: fail on arch a_sse4_1
1: offset 8630 in1: -3.32912e-05 in2: -3.31416e-05 tolerance was: 0.001
1: offset 14489 in1: 15898.6 in2: 15917.7 tolerance was: 0.001
1: offset 28068 in1: 9.8141e-05 in2: 9.79914e-05 tolerance was: 0.001
1: offset 31758 in1: -0.000106247 in2: -0.000106098 tolerance was: 0.001
1: offset 44343 in1: 189075 in2: 186414 tolerance was: 0.001
1: offset 93063 in1: -6.97692e-05 in2: -6.96196e-05 tolerance was: 0.001
1: volk_32f_tan_32f: fail on arch u_sse4_1
1: Best aligned arch: a_sse4_1
1: Best unaligned arch: u_sse4_1

n-west, 2016-01-12:
This is a side-effect of actually fixing the implementations (https://github.com/gnuradio/volk/issues/52#issuecomment-161068083). Some branch merging accidentally overwrote the fix. It should no longer fail QA, but kernel behavior has not changed

closed 2016-01-12 by n-west