gnuradio/volk #63: __cpuid_count() for MSVC

gnieboer, 2016-02-06:
Volk won't build on MSVC because __cpuid_count() isn't available.  This was posted previously to the discussion board but the submitter never submitted an issue here.  It's cut and pasted below...

-------------------
I noticed since last time (a year ago) I tried building
Volk using MSVC v16, the volk_cpu.tmp.c now uses the
gcc-centric function '__cpuid_count()' which MSVC doesn't
have.

I'm not really sure if the below patch is correct. But I
assume (from looking at gcc 5.1's source) that the ECX should
be loaded with the 'count' value. So could it be patched into
something like this?

--- a/volk/tmpl/volk_cpu.tmpl.c  2015-09-01 13:52:53
+++ b/volk_cpu.tmpl.c     2015-09-07 13:44:25
@@ -71,8 +71,16 @@

 static inline unsigned int cpuid_count_x86_bit(unsigned int level, unsigned int count, unsigned int reg, unsigned int bit) {
 #if defined(VOLK_CPU_x86)
+    #if defined(_MSC_VER) && defined(HAVE_INTRIN_H)
+      int regs[4];
+      __cpuidex(regs, level, count);
+    #elif defined(__GNUC__)
     unsigned int regs[4];
     __cpuid_count(level, count, regs[0],  regs[1],  regs[2], regs[3]);
+    #else
+      #error No __cpuid()!
+    #endif
     return regs[reg] >> bit & 0x01;
 #else
     return 0;

-------

Just to let you know.

The docs on MSVC's __cpuidex() is here:
  https://msdn.microsoft.com/en-us/library/vstudio/hskdteyh(v=vs.100).aspx

-- 
--gv

guruofquality, 2016-02-06:
FWIW, I have been keeping around a similar patch for my windows build: https://github.com/pothosware/gnuradio/issues/17

n-west, 2016-02-07:
It looks like you have functionally equivalent patches, and I'm going to take a version of Josh's that works with gcc. Confirmation that this fixes your windows issues would be nice as I have no way to test windows stuff

guruofquality, 2016-02-07:
Thanks I appreciate it. While we are on the subject:

* I have [this compiler name patch](https://github.com/pothosware/PothosSDR/blob/master/patches/volk_fix_msvc14.diff) as well to set the "cmake_c_compiler_version" for vc14. Provided by user @bkerler
* And [added config.h idiom support to qa_utils.cc](https://github.com/pothosware/PothosSDR/blob/master/patches/volk_qa_utils_config_h.diff) which needed it for the M_PI definition.

These are all patched off of the latest release tag and they are very minor. I also have some vc11 patches, but vc11 is quite old and I am hoping to eventually drop support for it -- and so I wouldn’t bother with them. So anyway, maybe for the next volk release I will be able to build stock volk patch free -- and thats pretty cool. Thanks again.

n-west, 2016-02-07:
The M_PI definition is no longer there and 99.9% sure it won't be coming back, so I'll skip that one, but adding VC 14 is no problem. You're also correct about VC11. I'll need to figure out a good time to remove the similarly ancient VCs. Probably around the same time I switch from cheetah to mako and call it 3.0

closed 2016-02-07 by n-west