gnuradio/volk #66: Debug message contaminates standard out

vsergeev, 2016-02-27:
The "Using Volk machine:" message https://github.com/gnuradio/volk/blob/v1.2.1/tmpl/volk.tmpl.c#L56 contaminates stdout, which interferes with programs that are trying to use that as a channel for structured data (e.g. JSON, etc.).

Could this message be moved to standard error, or even omitted altogether, since `volk_get_machine()` already provides this information in the public API?

vsergeev, 2016-04-08:
#48 or similar would address this.

n-west, 2016-04-11:
I'm on board with directing messages like this somewhere other than stdout by default, but don't think that PR is the right approach. I'd like to see a very lightweight logging framework, probably compile time configurable.

vsergeev, 2016-04-11:
Sounds good to me. At the moment (v1.2.2), it seems that volk only has only two active call sites to `printf()` -- `get_machine()` (https://github.com/gnuradio/volk/blob/v1.2.2/tmpl/volk.tmpl.c#L35), and `volk_get_index()` (https://github.com/gnuradio/volk/blob/v1.2.2/lib/volk_rank_archs.c#L41) -- that are triggered through use of the library. In the meanwhile, wouldn't it be prudent to move these messages (4 lines of code) to stderr, or even omit them altogether, while the better logging approach is being developed? I can provide a PR if this is reasonable.

PR #75 (2016-04-13): Silence debug messages on standard out
The first commit comments out the "Using Volk machine: <machine>" debug
message on standard out, which interferes with programs trying to use
standard out for legitimate purposes.

This modification is consistent with other commented out debug printf()
calls in the volk codebase, and should be superseded by a proper
compile-time configurable logging system for the library.

The second commit moves the "no arch found" warning message from stdout to stderr, thereby eliminating all library output on stdout.

This PR fixes #66.
closed 2016-04-13 by n-west