gnuradio/volk #70: volk_32f_x2_dot_prod_32f missing MSVC compatible aligned var declaration when using AVX2

gnieboer, 2016-03-16:
MSVC uses a different syntax for aligned data types, so when AVX2 is enabled, compilation will fail.  

See this [commit](https://github.com/gnuradio/volk/pull/68/commits/3536338cd901e9fab4cf1bf5ac207f95b9edfccd) in [this](https://github.com/gnuradio/volk/pull/68) pull request for a patch that conditionally compiles the appropriate syntax.

Thought... if we expect introduction of a lot more aligned data types, you may want to define a generic macro in a header to avoid cluttering the code.  
#ifdef _MSC_VER
#define ALIGNED(WIDTH, VAR) _declspec(align(WIDTH)) VAR
#else 
#define ALIGNED(WIDTH, VAR) VAR__attribute__((aligned(WIDTH)))
#endif

guruofquality, 2016-03-16:
@gnieboer if you need alignment, use the __VOLK_ATTR_ALIGNED() macro from https://github.com/gnuradio/volk/blob/master/include/volk/volk_common.h

Use the __VOLK_ATTR_ALIGNED() macro before the declaration, and not at the end. Then it will work for both compilers with no ifdef

gnieboer, 2016-03-16:
OK, didn't see there was already a macro in there.  Then definitely disregard my fix and simply replace said code with the macro.

guruofquality, 2016-03-16:
@gnieboer I was just trying to help out and keep the patches clean and neat -- and hopefully easier for the maintainer to merge :-) I don't have a way to upstream the patch myself, so you should probably make the ```__VOLK_ATTR_ALIGNED``` change part of your pull request.

gnieboer, 2016-03-16:
Agreed, clean is good.  Change made to pull request!

n-west, 2017-02-24:
Is this still a problem? I think the appropriate PR was merged.

closed 2017-02-24 by n-west