gnuradio/volk #71: enable better arch detection in MSVC compiler / 64-bit

gnieboer, 2016-03-16:
When compiling for MSVC, only the SSE machine is detected.  This is due to a couple reasons:

1- The xgetbv test won't compile on MSVC because an intrinsic already exists.  [This commit](https://github.com/gnuradio/volk/pull/68/commits/536f51a85e3bcbd87146a9ac83a5895e02646d18) in [this pull request](https://github.com/gnuradio/volk/pull/68) will patch that by adding a different test when compiling for MSVC (the test also checks if the CPU and OS support the instruction).

2- The cvtpi32_ps instrinsic isn't available on 64-bit machines, and in addition to marking it as unavailable, it marks AVX as a whole unavailable.   [This commit](https://github.com/gnuradio/volk/pull/68/commits/536f51a85e3bcbd87146a9ac83a5895e02646d18) in [this pull request](https://github.com/gnuradio/volk/pull/68) will patch that by testing for 64-bit compilation (on any compiler) and if 64-bit, marking the instruction as unavailable but leaving AVX available.

3- archs.xml only checks for FMA support if GCC.[This commit](https://github.com/gnuradio/volk/pull/68/commits/eec9b5f453c057f56b1e4327d887e7fcca04c041) will add a check for an MSVC that supports FMA (MSVC 2015 does as part of the AVX2 support)

4- machines.xml does not allow 64-bit support for SSE3/4.  [This commit](https://github.com/gnuradio/volk/pull/68/commits/f8b30174eb812aaae319ac7115c4f91ae3264583) enables it.

With the above patches, MSVC compiled fine on a CPU w/ AVX2+FMA support, and volk_profile ran successfully.  There were 2-3 profile tests that failed, but volk_profile continued and appeared to discount those results and choose different arches.  I'll post details of those errors in a different issue.

gnieboer, 2016-03-16:
The three tests that failed already have issues open, two for MSVC, and the third was reported for OS X Yosemite but appears to be nearly identical so I just added a comment there.

n-west, 2016-08-04:
@gnieboer can I get an update on this? I think relevant patches were merged

gnieboer, 2016-08-10:
Yes, all issues referenced here have been merged and will build.  Closing.

closed 2016-08-10 by gnieboer