gnuradio/volk #88: volk_free causes Segmentation fault

jdemel, 2016-06-01:
I generated a new GR block in an OOT module which is a 'noblock'.
I instantiate it inside a sync block with a boost::shared_ptr, do signal processing and destruct the flowgraph.
OS: Ubuntu 16.04 in a VirtualBox
VOLK version: 1.2.2, commit f1c4b5534db875a05e71b3b0dd3fbf486ef2c6dc

During destruction, I get a SegFault. That's the gdb output
------------
Thread 1 "python" received signal SIGSEGV, Segmentation fault.
0x00007ffff7870bd6 in _int_free (av=0x7ffff7bb4b20 <main_arena>, p=<optimized out>, have_lock=0)
    at malloc.c:4015
4015	malloc.c: No such file or directory.
------------

First I thought it might be a general problem. But that's not the case. I can 'volk_free' ~4 aligned arrays. (I do have 9 in total). But if I try to properly free 5 or more, I get this SegFault. I hope it helps to track this thing down.

That's the backtrace
--------------------
(gdb) bt
#0  0x00007ffff7870bd6 in _int_free (av=0x7ffff7bb4b20 <main_arena>, p=<optimized out>, have_lock=0)
    at malloc.c:4015
#1  0x00007ffff7874abc in __GI___libc_free (mem=<optimized out>) at malloc.c:2969
#2  0x00007fffd4f5e2a0 in gr::gfdm::modulator_kernel_cc::~modulator_kernel_cc() ()
   from [HOME]/src/gr-gfdm/build/lib/libgnuradio-gfdm.so
#3  0x00007fffd4f5ddab in void boost::checked_delete<gr::gfdm::modulator_kernel_cc>(gr::gfdm::modulator_kernel_cc*) () from [HOME]/src/gr-gfdm/build/lib/libgnuradio-gfdm.so
#4  0x00007fffd4f5de3c in boost::detail::sp_counted_impl_p<gr::gfdm::modulator_kernel_cc>::dispose() ()
   from [HOME]/src/gr-gfdm/build/lib/libgnuradio-gfdm.so
#5  0x00007fffd4f44ad2 in boost::detail::sp_counted_base::release() ()
   from [HOME]/src/gr-gfdm/build/lib/libgnuradio-gfdm.so
#6  0x00007fffd4f44b65 in boost::detail::shared_count::~shared_count() ()
   from [HOME]/src/gr-gfdm/build/lib/libgnuradio-gfdm.so
#7  0x00007fffd4f5d862 in boost::shared_ptr<gr::gfdm::modulator_kernel_cc>::~shared_ptr() ()
   from [HOME]/src/gr-gfdm/build/lib/libgnuradio-gfdm.so
#8  0x00007fffd4f5d3c1 in gr::gfdm::simple_modulator_cc_impl::~simple_modulator_cc_impl() ()
   from [HOME]/src/gr-gfdm/build/lib/libgnuradio-gfdm.so
#9  0x00007fffd4f5d474 in gr::gfdm::simple_modulator_cc_impl::~simple_modulator_cc_impl() ()
   from [HOME]/src/gr-gfdm/build/lib/libgnuradio-gfdm.so
#10 0x00007fffe984e1d2 in boost::detail::sp_counted_base::release (this=0x269bc40)
    at /usr/include/boost/smart_ptr/detail/sp_counted_base_gcc_x86.hpp:146
#11 0x00007fffe984e83c in std::_Destroy<gr::edge> (__pointer=<optimized out>)
    at /usr/include/c++/5/bits/stl_construct.h:93
#12 std::_Destroy_aux<false>::__destroy<gr::edge*> (__last=<optimized out>, __first=0x269b350)
    at /usr/include/c++/5/bits/stl_construct.h:103
#13 std::_Destroy<gr::edge*> (__last=<optimized out>, __first=<optimized out>)
    at /usr/include/c++/5/bits/stl_construct.h:126
#14 std::_Destroy<gr::edge*, gr::edge> (__last=0x269b380, __first=0x269b320)
    at /usr/include/c++/5/bits/stl_construct.h:151
#15 std::vector<gr::edge, std::allocator<gr::edge> >::_M_erase_at_end (this=0x2636010, __pos=0x269b320)
    at /usr/include/c++/5/bits/stl_vector.h:1438
#16 std::vector<gr::edge, std::allocator<gr::edge> >::clear (this=0x2636010)
    at /usr/include/c++/5/bits/stl_vector.h:1212
#17 gr::flowgraph::clear (this=0x2635ff0)
    at [prefix]/src/gnuradio/gnuradio-runtime/lib/flowgraph.cc:128
#18 0x00007fffe985f595 in gr::hier_block2_detail::disconnect_all (this=0x1fa3b90)
    at [prefix]/src/gnuradio/gnuradio-runtime/lib/hier_block2_detail.cc:518
#19 0x00007fffe985b0dc in gr::hier_block2::disconnect_all (this=this@entry=0x2638140)
    at [prefix]/src/gnuradio/gnuradio-runtime/lib/hier_block2.cc:140
#20 0x00007fffe985b0fc in gr::hier_block2::~hier_block2 (this=0x2638140, __in_chrg=<optimized out>)
    at [prefix]/src/gnuradio/gnuradio-runtime/lib/hier_block2.cc:60
#21 0x00007fffe98853b9 in gr::top_block::~top_block (this=0x2638140, __in_chrg=<optimized out>)
    at [prefix]/src/gnuradio/gnuradio-runtime/lib/top_block.cc:57
#22 0x00007fffea13313a in boost::detail::sp_counted_base::release (this=0x2515840)
    at /usr/include/boost/smart_ptr/detail/sp_counted_base_gcc_x86.hpp:146
#23 0x00007fffea135955 in boost::detail::sp_counted_base::release (this=<optimized out>)
    at [prefix]/src/gnuradio/build/gnuradio-runtime/swig/runtime_swigPYTHON_wrap.cxx:46736
#24 boost::detail::shared_count::~shared_count (this=0x2553b68, __in_chrg=<optimized out>)
    at /usr/include/boost/smart_ptr/detail/shared_count.hpp:443
#25 boost::shared_ptr<gr::top_block>::~shared_ptr (this=0x2553b60, __in_chrg=<optimized out>)
    at /usr/include/boost/smart_ptr/shared_ptr.hpp:323
#26 _wrap_delete_top_block_sptr (args=<optimized out>)
    at [prefix]/src/gnuradio/build/gnuradio-runtime/swig/runtime_swigPYTHON_wrap.cxx:46723
#27 0x00000000004b1153 in PyObject_Call ()
#28 0x00000000004b9ee2 in PyObject_CallFunctionObjArgs ()
#29 0x00007fffea1008a5 in SwigPyObject_dealloc (v=0x7fffc8fbfcc0)
    at [prefix]/src/gnuradio/build/gnuradio-runtime/swig/runtime_swigPYTHON_wrap.cxx:1677
#30 0x00000000004fd7c6 in ?? ()
#31 0x00000000004fd7c6 in ?? ()
#32 0x00000000004a0d42 in PyDict_SetItem ()
#33 0x00000000004dad6d in _PyObject_GenericSetAttrWithDict ()
#34 0x00000000004da8eb in PyObject_SetAttr ()
#35 0x00000000004c588a in PyEval_EvalFrameEx ()
#36 0x00000000004ca39f in PyEval_EvalFrameEx ()
------------------------

jdemel, 2016-06-01:
A few more observations:
It seems like 2 different errors may be raised. Either SIGSEGV in _int_free at malloc.c:4015, or SIGABRT in __GI_raise at sysv/linux/raise.c:54

They both occur on the line of code which says 'volk_free(...)'. I checked and the array in question does not get free'd twice or such. Also, it would happen with the next array if I commented this one out.

This whole error only occurs during Python unittests. Not while the flowgraph is running normally and then exited.

jdemel, 2016-06-03:
It seems like I misused volk_malloc after all.

closed 2016-06-03 by jdemel

marcusmueller, 2017-08-09:
@jdemel , your lack of formatting code as code has led to your `#frame number` to reference issue/PR 1–36 in this PR. That's a bit of a nuisance – could you edit your backtrace and use the `<>` code formatting, to avoid this