gnuradio/volk #92: build issues with c++98 and gcc4 (libboost1.58 with c++11 style symbols)

osh, 2016-06-06:
When building with gcc-4.x 
it is forced into c++98 mode resulting in undefined c++11 style ABI symbols not found when linking against libboost built for c++11.   Using gcc-5.x fixes the issue ... 
however 
by propperly setting 
_GLIBCXX_USE_CXX11_ABI 
somewhere, this should be fixable 
results in a non-helpful error message about simply missing symbols when linking the first binary (volk_profile)

n-west, 2016-06-15:
woah. What system/distro is this easily reproducible on?

n-west, 2016-06-30:
@tim is this still a thing?

osh, 2016-06-30:
yes

n-west, 2016-07-02:
I think this is fixed just before I tagged 1.3 release, but cannot reproduce

n-west, 2016-07-13:
I am now running in to this issue and my added define doesn't fix it. Have you gotten this to work? I'm not sure it's as simple as adding the define.

https://lwn.net/Articles/632658/

I'm thinking that BOOST should have been built with this flag set to 0 until Ubuntu completely removes support for gcc 4, which makes this somewhat of a packaging bug.

n-west, 2016-07-13:
This cannot be fixed in VOLK. We simply cannot link against a boost with CXX11 ABI.

closed 2016-07-13 by n-west