gnuradio/volk #93: Missing include files

slarew, 2016-07-14:
I found that my installation of volk (v1.3) was missing some include files. For example, `volk/saturation_arithmetic.h` is not installed.  Glancing at the CMakeLists.txt code, I noticed that `volk/volk_sse_intrinsics.h` also seems to be missing.  Maybe there are more that I missed.

n-west, 2016-08-04:
Hi-- do you actually need these files? They should only be required to build the lib and it would be a little odd to have user-code using them. If you have a valid usecase, for example a volk OOT, then I'm happy to accept a PR

slarew, 2016-08-04:
I found that `volk/saturation_arithmetic.h` is included by the public header `volk/volk_16ic_x2_dot_prod_16ic.h`. Also, `volk/volk_32fc_magnitude_32f.h` includes `volk/volk_sse_intrinsics.h`.

FYI, I encountered this issue while trying to create a sort of convenience umbrella header that includes gnuradio, volk, pmt headers.

n-west, 2017-08-14:
I included those files for installation.

closed 2017-08-14 by n-west