gnuradio/volk #97: __VOLK_PREFETCH() macro for __builtin_prefetch

guruofquality, 2016-07-20:
So I have been tracking maint and building PothosSDR installers with gnuradio/volk under visual studio. Anyway, I ran into this and finally went after it today to get things compiling again. The __builtin_prefetch() isnt very portable so I created a __VOLK_PREFETCH() macro in the style of the other similar cross platform macros.

This is the only important change to volk_common.h, the [rest of the changes](https://github.com/pothosware/PothosSDR/blob/master/patches/volk_prefetch_compat_macro.diff) are just a matter of sed replacing the builtin with the macro. Didnt implement the prefetch for msvc, this is just a NOP that gets it compiling and should be harmless:
```
diff --git a/include/volk/volk_common.h b/include/volk/volk_common.h
index 4d35f5c..a53b139 100644
--- a/include/volk/volk_common.h
+++ b/include/volk/volk_common.h
@@ -16,6 +16,7 @@
 #    define __VOLK_ATTR_EXPORT
 #    define __VOLK_ATTR_IMPORT
 #  endif
+#  define __VOLK_PREFETCH(addr)  __builtin_prefetch(addr)
 #elif _MSC_VER
 #  define __VOLK_ATTR_ALIGNED(x) __declspec(align(x))
 #  define __VOLK_ATTR_UNUSED
@@ -23,6 +24,7 @@
 #  define __VOLK_ATTR_DEPRECATED __declspec(deprecated)
 #  define __VOLK_ATTR_EXPORT     __declspec(dllexport)
 #  define __VOLK_ATTR_IMPORT     __declspec(dllimport)
+#  define __VOLK_PREFETCH(addr)
 #else
 #  define __VOLK_ATTR_ALIGNED(x)
 #  define __VOLK_ATTR_UNUSED
@@ -30,6 +32,7 @@
 #  define __VOLK_ATTR_DEPRECATED
 #  define __VOLK_ATTR_EXPORT
 #  define __VOLK_ATTR_IMPORT
+#  define __VOLK_PREFETCH(addr)
#endif
```

So I am hoping that this is a nice portable way to handle the prefetch that in the style of volk and can go into maint. Otherwise, a simple ``#define __builtin_prefetch(addr)`` would be a easy one line patch for me to carry around instead; it would also do the same trick, albeit less proper I feel. Thanks for your consideration!

gnieboer, 2016-08-10:
Agreed, won't compile under MSVC without defining as a null at least.

Previously, __builtin_prefetch was only used for NEON arches it looks like, but now
volk_16ic_x2_dot_prod_16ic.h
volk_32fc_convert_16ic.h
will include them in non-NEON arches.

n-west, 2017-02-24:
Closing because the merged issue should fix it. Let me know if this is still a problem

closed 2017-02-24 by n-west