gnuradio/volk #110: Make error: processor does not support ARM mode `sbfx r11,r1,#2,#1'

MiBatu, 2017-02-26:
Fail to install on RPi3 (Debian):

Building ASM object lib/CMakeFiles/volk_obj.dir/__/kernels/volk/asm/neon/volk_32f_x2_dot_prod_32f_neonasm_opts.s.o
/home/pi/gnuradio/volk/kernels/volk/asm/neon/volk_32f_x2_dot_prod_32f_neonasm_opts.s: Assembler messages:
/home/pi/gnuradio/volk/kernels/volk/asm/neon/volk_32f_x2_dot_prod_32f_neonasm_opts.s:46: Error: selected processor does not support ARM mode `sbfx r11,r1,#2,#1'
lib/CMakeFiles/volk_obj.dir/build.make:1462: recipe for target 'lib/CMakeFiles/volk_obj.dir/__/kernels/volk/asm/neon/volk_32f_x2_dot_prod_32f_neonasm_opts.s.o' failed
make[2]: *** [lib/CMakeFiles/volk_obj.dir/__/kernels/volk/asm/neon/volk_32f_x2_dot_prod_32f_neonasm_opts.s.o] Error 1
CMakeFiles/Makefile2:195: recipe for target 'lib/CMakeFiles/volk_obj.dir/all' failed
make[1]: *** [lib/CMakeFiles/volk_obj.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2

n-west, 2017-02-27:
See this blog post:

http://lukeberndt.com/2016/compiling-gnuradio-on-a-raspberry-pi/

The raspberry pi OS is built with all kinds of strange flags that make detection hard.

MiBatu, 2017-03-06:
Thank you, but this link doesn't work =(

"Error establishing a database connection"

MiBatu, 2017-03-06:
I've pushed 'make' to go forward by 'make -B -i' and there is some other warnings:

_[ 72%] Building C object lib/CMakeFiles/volk_obj.dir/volk_machine_neon_hardfp.c.o
In file included from /home/pi/volk/build/lib/volk_machine_neon_hardfp.c:113:0:
/home/pi/volk/kernels/volk/volk_32fc_s32fc_multiply_32fc.h: In function ‘volk_32fc_s32fc_multiply_32fc_neon’:
/home/pi/volk/kernels/volk/volk_32fc_s32fc_multiply_32fc.h:282:16: warning: unused variable ‘cPtr’ [-Wunused-variable]
     lv_32fc_t* cPtr = cVector;
                ^
In file included from /home/pi/volk/build/lib/volk_machine_neon_hardfp.c:97:0:
/home/pi/volk/build/lib/volk_machine_neon_hardfp.c: In function ‘volk_16i_max_star_horizontal_16i_neon’:
/home/pi/volk/kernels/volk/volk_16i_max_star_horizontal_16i.h:169:9: warning: ‘zeros’ is used uninitialized in this function [-Wuninitialized]
   zeros = veorq_s16(zeros, zeros);
         ^
In file included from /home/pi/volk/build/lib/volk_machine_neon_hardfp.c:159:0:
/home/pi/volk/build/lib/volk_machine_neon_hardfp.c: In function ‘volk_16i_x4_quad_max_star_16i_neon’:
/home/pi/volk/kernels/volk/volk_16i_x4_quad_max_star_16i.h:221:9: warning: ‘zeros’ is used uninitialized in this function [-Wuninitialized]
   zeros = veorq_s16(zeros, zeros);
         ^
In file included from /home/pi/volk/build/lib/volk_machine_neon_hardfp.c:131:0:
/home/pi/volk/build/lib/volk_machine_neon_hardfp.c: In function ‘volk_16i_max_star_16i_neon’:
/home/pi/volk/kernels/volk/volk_16i_max_star_16i.h:132:9: warning: ‘zeros’ is used uninitialized in this function [-Wuninitialized]
   zeros = veorq_s16(zeros, zeros);
         ^
[ 72%] Built target volk_obj
Scanning dependencies of target volk
make[2]: *** No rule to make target 'lib/CMakeFiles/volk_obj.dir/__/kernels/volk/asm/neon/volk_32f_x2_dot_prod_32f_neonasm_opts.s.o', needed by 'lib/libvolk.so.1.3.0git'.  Stop.
[ 75%] Built target volk
Scanning dependencies of target test_all
[ 77%] Building CXX object lib/CMakeFiles/test_all.dir/testqa.cc.o
[ 79%] Building CXX object lib/CMakeFiles/test_all.dir/qa_utils.cc.o
make[2]: *** No rule to make target 'lib/libvolk.so.1.3.0git', needed by 'lib/test_all'.  Stop.
[ 81%] Built target test_all
Scanning dependencies of target volk-config-info
[ 84%] Building CXX object apps/CMakeFiles/volk-config-info.dir/volk-config-info.cc.o
make[2]: *** No rule to make target 'lib/libvolk.so.1.3.0git', needed by 'apps/volk-config-info'.  Stop.
[ 86%] Built target volk-config-info
Scanning dependencies of target volk_profile
[ 88%] Building CXX object apps/CMakeFiles/volk_profile.dir/volk_profile.cc.o
[ 90%] Building CXX object apps/CMakeFiles/volk_profile.dir/__/lib/qa_utils.cc.o
make[2]: *** No rule to make target 'lib/libvolk.so.1.3.0git', needed by 'apps/volk_profile'.  Stop._

Are they related to flag detection?

MiBatu, 2017-03-06:
Direct settings hepled to install VOLK =)
Command:

cmake cmake -Dhave_mfpu_neon=0 -DCMAKE_CXX_FLAGS:STRING="-march=armv6 -mfpu=vfp -mfloat-abi=hard" -DCMAKE_C_FLAGS:STRING="-march=armv6 -mfpu=vfp -mfloat-abi=hard" ..

BUT gnuradio still can't find it:

> -- Configuring VOLK support...
> --   VOLK submodule is not checked out.
> --   To check out the VOLK submodule, use:
> --     git pull --recurse-submodules=on
> --     git submodule update
> --   External VOLK disabled.
> --   Override with -DENABLE_INTERNAL_VOLK=ON/OFF
> -- 
> CMake Error at CMakeLists.txt:362 (message):
>   VOLK required but not found.

n-west, 2017-03-07:
The message indicates that the volk submodule in gnuradio is not checked out, which indicates you have built VOLK outside of GNU Radio. Have you tried using the cmake flag that the error message suggests, -DENABLE_INTERNAL_VOLK=OFF?

MiBatu, 2017-03-08:
Yes, and it was helpful to **make** GNU Radio without VOLK. But after **make install**  I do not see it correctly placed in X, as it usually happen when I use **$ apt-get install gnuradio**... so now looking how to run GNU Radio I've installed...

PS: will try to rebuild VOLK inside gnuradio dir. Thank you for your note.

closed 2017-03-13 by n-west