gnuradio/volk #114: SIGBUS Crash on ARMv7l with NEON

ddcc, 2017-04-24:
On a ODROID-XU4 ARM v7l machine, using the VOLK `neon_hardfp_orc` machine profile, I'm getting SIGBUS exceptions with certain neonasm kernels. These seem to be occurring because the input data isn't properly aligned, for example, with [`volk_32f_x2_add_32f_neonasm+24`](https://github.com/gnuradio/volk/blob/master/kernels/volk/asm/neon/volk_32f_x2_add_32f_neonasm.s#L30), the `vld1.32` instruction has the `:128` alignment specifier, but the input is clearly not aligned, as shown below:

Assembly:
```
0xb5b6d9e0 <volk_32f_x2_add_32f_neonasm>      push   {r7, r8, r10}
0xb5b6d9e4 <volk_32f_x2_add_32f_neonasm+4>    lsrs   r7, r3, #2
0xb5b6d9e8 <volk_32f_x2_add_32f_neonasm+8>    beq    0xb5b6da18 <volk_32f_x2_add_32f_neonasm+56>
0xb5b6d9ec <volk_32f_x2_add_32f_neonasm+12>   mov    r8, #0
0xb5b6d9f0 <volk_32f_x2_add_32f_neonasm+16>   pld    [r1, #128]      ; 0x80
0xb5b6d9f4 <volk_32f_x2_add_32f_neonasm+20>   pld    [r2, #128]      ; 0x80
0xb5b6d9f8 <volk_32f_x2_add_32f_neonasm+24>   vld1.32        {d0-d1}, [r1 :128]!
0xb5b6d9fc <volk_32f_x2_add_32f_neonasm+28>   add    r8, r8, #1
0xb5b6da00 <volk_32f_x2_add_32f_neonasm+32>   vld1.32        {d2-d3}, [r2 :128]!
0xb5b6da04 <volk_32f_x2_add_32f_neonasm+36>   vadd.f32       q2, q1, q0
0xb5b6da08 <volk_32f_x2_add_32f_neonasm+40>   cmp    r8, r7
0xb5b6da0c <volk_32f_x2_add_32f_neonasm+44>   vst1.32        {d4-d5}, [r0 :128]!
```

Registers:
```
r0             0x9ca5cbe4       2628111332
r1             0x9ca5cbe4       2628111332
r2             0x9ca9cbe4       2628373476
r3             0x13     19
```

Call Stack:
```
#0  0xb5b6d9fc in volk_32f_x2_add_32f_neonasm () from /usr/lib/arm-linux-gnueabihf/libvolk.so.1.1
#1  0xb6c0c70e in gr::blocks::add_ff_impl::work(int, std::vector<void const*, std::allocator<void const*> >&, std::vector<void*, std
::allocator<void*> >&) () from /usr/lib/arm-linux-gnueabihf/libgnuradio-blocks.so.3.7.9
#2  0xb6e2dca2 in gr::sync_block::general_work(int, std::vector<int, std::allocator<int> >&, std::vector<void const*, std::allocator
<void const*> >&, std::vector<void*, std::allocator<void*> >&) () from /usr/lib/arm-linux-gnueabihf/libgnuradio-runtime.so.3.7.9
#3  0xb6e0001a in gr::block_executor::run_one_iteration() () from /usr/lib/arm-linux-gnueabihf/libgnuradio-runtime.so.3.7.9
#4  0xb6e356b8 in gr::tpb_thread_body::tpb_thread_body(boost::shared_ptr<gr::block>, int) ()
   from /usr/lib/arm-linux-gnueabihf/libgnuradio-runtime.so.3.7.9
```

Looking through the other NEON kernels, it seems that others may have the same problem, e.g. `volk_32f_x2_dot_prod_32f_neonasm` and `volk_32f_s32f_multiply_32f_neonasm`. There was also a similar issue reported in #101. Since I'm not familiar with the VOLK code, I don't know if this is actually a bug in VOLK, or in GNU Radio, since the vectors in `gr::block_executor` are not allocated on the heap using `volk_malloc()` and `volk_get_alignment()`. But for now, I have patched the binary by removing the alignment specifiers in this function to avoid the bug.

n-west, 2017-04-24:
I'm not convinced this is the cause of that bug. Are you able to show that SIGBUS occurs before but not after this change? If I remember correctly, the 128 here isn't alignment and NEON assembly doesn't have an alignment specifier. The 128 here says to load 128 bits.

ddcc, 2017-04-24:
Yes, I patched the binary by eliminating the three alignment specifiers in  volk_32f_x2_add_32f_neonasm, and the SIGBUS went away. The other *_neonasm functions have the same problem, but I don't believe they're being called. This is a common problem, see e.g. https://mailman.videolan.org/pipermail/x264-devel/2015-December/011518.html .

n-west, 2017-04-24:
Hmm, ok. I missed this years ago. I think the right thing to do is have unaligned and aligned versions for all of the protokernels in question.

n-west, 2017-08-14:
I think this should be fixed now.

closed 2017-08-14 by n-west