gnuradio/volk #118: Problem adding new proto-kernel to volk

dmiralles2009, 2017-08-05:
Hey everyone:
I am trying to add a couple of proto kernel to the volk libraries that extend to AVX2 intrinsics. I have been able to add the code and compilation goes fine. However, when running the `volk_profile` executable after compilation my new proto-kernels are not added to the profiling test some how. See below:
RUN_VOLK_TESTS: volk_32f_s32f_convert_8i(131071,1987)
u_sse2 completed in 86.654ms
u_sse completed in 370.457ms
generic completed in 1873.98ms
a_sse2 completed in 69.092ms
a_sse completed in 354.4ms
a_generic completed in 1833.29ms
Best aligned arch: a_sse2
Best unaligned arch: u_sse2
 Note that I am only adding code to existing proto kernel, for example adding `volk_32f_s32f_convert_8i_u_avx2` to the file `volk_32f_s32f_convert_8i.h`. Code for this can be seen in the [fork](https://github.com/dmiralles2009/volk/tree/feature/adding_more_avx2_intrinsics) I recently made 

In addition, I have been able to add protokernels to other volk modules like the ones used in the [GNSS-SDR](https://github.com/gnss-sdr/gnss-sdr/tree/master/src/algorithms/libs/volk_gnsssdr_module/volk_gnsssdr) software. Not sure why the process is not functional here.

If this is not an issue or I am posting on the wrong forum my apologies in advance, however, help is more than welcome.
Thanks in advance

n-west, 2017-08-05:
Can you rerun cmake? There's some code and templates that get generated when you run cmake that are necessary for new protokernels to work

dmiralles2009, 2017-08-05:
I have done it multiple occasions and still does not seems to work as expected. Below the output of cmake
```
dmiralles@dmiralles-Q400A:~/Documents/volk/build2$ cmake ../
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Build type not specified: defaulting to release.
-- Build type set to Release.
-- Found Git: /usr/bin/git (found version "2.7.4") 
-- Extracting version information from git describe...
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.12", minimum required is "2") 
-- 
-- Python checking for python >= 2.7
-- Python checking for python >= 2.7 - found
-- 
-- Python checking for mako >= 0.4.2
-- Python checking for mako >= 0.4.2 - found
-- 
-- Python checking for six - python 2 and 3 compatibility library
-- Python checking for six - python 2 and 3 compatibility library - found
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   filesystem
--   system
--   unit_test_framework
--   program_options
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'orc-0.4 > 0.4.11'
--   No package 'orc-0.4' found
-- orc files (missing:  ORC_LIBRARY ORC_INCLUDE_DIR ORCC_EXECUTABLE) 
-- Found Doxygen: /usr/bin/doxygen (found version "1.8.11") 
-- QA Testing is enabled.
--   Modify using: -DENABLE_TESTING=ON/OFF
-- System profiling is disabled.
--   Modify using: -DENABLE_PROFILING=ON/OFF
-- Looking for cpuid.h
-- Looking for cpuid.h - found
-- Looking for intrin.h
-- Looking for intrin.h - not found
-- Looking for fenv.h
-- Looking for fenv.h - found
-- Looking for dlfcn.h
-- Looking for dlfcn.h - found
-- Compiler name: GNU
-- Performing Test HAVE_WERROR_UNUSED_CMD_LINE_ARG
-- Performing Test HAVE_WERROR_UNUSED_CMD_LINE_ARG - Failed
-- Looking for posix_memalign
-- Looking for posix_memalign - found
-- x86* CPU detected
-- Performing Test have_mfloat_abi_softfp
-- Performing Test have_mfloat_abi_softfp - Failed
-- Performing Test have_mfloat_abi_hard
-- Performing Test have_mfloat_abi_hard - Failed
-- Performing Test have_mfpu_neon
-- Performing Test have_mfpu_neon - Failed
-- Performing Test have_funsafe_math_optimizations
-- Performing Test have_funsafe_math_optimizations - Success
-- Performing Test have_m32
-- Performing Test have_m32 - Failed
-- Performing Test have_m64
-- Performing Test have_m64 - Success
-- Performing Test have_m3dnow
-- Performing Test have_m3dnow - Success
-- Performing Test have_msse4_2
-- Performing Test have_msse4_2 - Success
-- Performing Test have_mpopcnt
-- Performing Test have_mpopcnt - Success
-- Performing Test have_mmmx
-- Performing Test have_mmmx - Success
-- Performing Test have_mfma
-- Performing Test have_mfma - Success
-- Performing Test have_msse
-- Performing Test have_msse - Success
-- Performing Test have_msse2
-- Performing Test have_msse2 - Success
-- Performing Test have_msse3
-- Performing Test have_msse3 - Success
-- Performing Test have_mssse3
-- Performing Test have_mssse3 - Success
-- Performing Test have_msse4a
-- Performing Test have_msse4a - Success
-- Performing Test have_msse4_1
-- Performing Test have_msse4_1 - Success
-- Performing Test have_mavx
-- Performing Test have_mavx - Success
-- Performing Test have_mavx2
-- Performing Test have_mavx2 - Success
-- ORC support not found, Overruled arch orc
-- Check size of void*[8]
-- Check size of void*[8] - done
-- CPU width is 64 bits, Overruled arch 32
-- Available architectures: generic;64;3dnow;abm;popcount;mmx;fma;sse;sse2;norc;sse3;ssse3;sse4_a;sse4_1;sse4_2;avx;avx2
-- Available machines: generic;sse2_64_mmx;sse3_64_mmx;ssse3_64_mmx;sse4_a_64_mmx;sse4_1_64_mmx;sse4_2_64_mmx;avx_64_mmx;avx2_64_mmx
-- BUILD TYPE = RELEASE
-- Base cflags = -O3 -DNDEBUG  -Wall
-- BUILD INFO ::: generic ::: GNU ::: -O3 -DNDEBUG  -Wall 
-- BUILD INFO ::: sse2_64_mmx ::: GNU ::: -O3 -DNDEBUG  -Wall -m64 -mmmx -msse -msse2
-- BUILD INFO ::: sse3_64_mmx ::: GNU ::: -O3 -DNDEBUG  -Wall -m64 -mmmx -msse -msse2 -msse3
-- BUILD INFO ::: ssse3_64_mmx ::: GNU ::: -O3 -DNDEBUG  -Wall -m64 -mmmx -msse -msse2 -msse3 -mssse3
-- BUILD INFO ::: sse4_a_64_mmx ::: GNU ::: -O3 -DNDEBUG  -Wall -m64 -mmmx -msse -msse2 -msse3 -msse4a -mpopcnt
-- BUILD INFO ::: sse4_1_64_mmx ::: GNU ::: -O3 -DNDEBUG  -Wall -m64 -mmmx -msse -msse2 -msse3 -mssse3 -msse4.1
-- BUILD INFO ::: sse4_2_64_mmx ::: GNU ::: -O3 -DNDEBUG  -Wall -m64 -mmmx -msse -msse2 -msse3 -mssse3 -msse4.1 -msse4.2 -mpopcnt
-- BUILD INFO ::: avx_64_mmx ::: GNU ::: -O3 -DNDEBUG  -Wall -m64 -mmmx -msse -msse2 -msse3 -mssse3 -msse4.1 -msse4.2 -mpopcnt -mavx
-- BUILD INFO ::: avx2_64_mmx ::: GNU ::: -O3 -DNDEBUG  -Wall -m64 -mmmx -msse -msse2 -msse3 -mssse3 -msse4.1 -msse4.2 -mpopcnt -mavx -mfma -mavx2
-- Compiler Version: cc (Ubuntu 5.4.0-6ubuntu1~16.04.4) 5.4.0 20160609
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- c flags:  -Wall;
-- asm flags: 
-- Did not find liborc and orcc, disabling orc support...
-- Loading version 1.3.0git into constants...
-- Using install prefix: /usr/local
-- Configuring done
-- Generating done
-- Build files have been written to: /home/dmiralles/Documents/volk/build2
```
Thanks for the help