gnuradio/volk #129: Add FMA operations

n-west, 2017-09-28:
AVX2/FMA support exists, but only two protokernels use it (volk_32f_x2_dot_prod_32f, volk_32fc_x2_multiply_32fc) because it was slower on new hardware in the first generation. Now, it appears FMA has advantages on the latest hardware. Several multiply and dot_prod operations might benefit from FMA operations, so create new protokernels where appropriate.

yashkar, 2017-09-30:
Hi,
   I was planning to work on the FPU unit. But before that, I plan to work on adding FMA operations to understand the vox environment. Can this work be assigned to me? Also, if you would like to give any suggestions as to how I should go about building this up, please let me know!

n-west, 2017-09-30:
Thanks for the interest. Hopefully you browsed through the code and website enough to understand the overall gist of VOLK. The easiest thing to do would be look at kernels/volk/volk_32fc_x2_multiply_32fc.h for the function ending with avx2_fma, then make a version for kernels/volk/volk_32fv_x2_multiply_conjugate_32fc.h. You'll only have to change enough to get a conjugate.

You can test that it works by running make test (or ctest) and grepping for the kernel you care about. Holler if you need any help or more direct pointers. I can also spend some time doing an example of one of the kernels that would be helpful to show you a rough template for how it can be done

yashkar, 2017-10-01:
Thanks for the reply! I did go through the code and the web description to get an understanding of VOLK. I'll go through the volk_32fc_x2_multiply_32fc.h and volk_32fv_x2_multiply_conjugate_32fc.h more closely over the next few days. I did look through the two codes and seems like I need to simply work on the conjugate for AVX2. I ll post after I look closely at the code in a few days.

yashkar, 2017-10-01:
I am trying to make the build dir. However, when trying to do so, I am getting a for loop error. The for loop initial declaration are not supported. I guess we need to change the cmake file to support C99 operations as well.

n-west, 2017-10-02:
I've seen that-- my guess is you are using GCC 4.8 or 4.9. I need to update the minimum cmake version or gcc version. Basically, set(CMAKE_C_STANDARD ${STANDARD}) was introduced in v3.1 (I think). A quick hack is to set(CMAKE_C_FLAGS ${CMAKE_C_FLAGS} c99) somewhere in top level cmakelists.txt.

closed 2018-02-22 by n-west