gnuradio/volk #130: volk_32f_s32f_mod_rangepuppet_32f missing

ast, 2017-10-26:
Build fails if I don't comment out line 133 in kernel_tests.h

```
uname -a
Darwin albinsmacbook-7.local 16.7.0 Darwin Kernel Version 16.7.0: Thu Jun 15 17:36:27 PDT 2017; root:xnu-3789.70.16~2/RELEASE_X86_64 x86_64 i386 MacBookPro11,1 Darwin

kernel_tests.h:133:10: error: use of undeclared identifier 'volk_32f_s32f_mod_rangepuppet_32f_get_func_desc'; did you mean
      'volk_32fc_s32fc_rotatorpuppet_32fc_get_func_desc'?
        (VOLK_INIT_TEST(volk_32f_s32f_mod_rangepuppet_32f,              test_params))
```

marcusmueller, 2018-02-01:
@n-west can you assign to me?

n-west, 2018-02-01:
I think you've already fixed in on master and I should've cherry-picked it
back to maint. If that hasn't happened yet, feel free

marcusmueller, 2018-02-11:
@n-west /me is confused; mod_range isn't even yet on maint! So, this isn't failing on maint, this is failing on some other branch / commit (@ast can you share your current git commit ID?).

So, I really don't know – maybe volk.h isn't generated correctly during the build? Lemme check whether I accidentally only define the puppet for one platform, though.

n-west, 2018-02-11:
Whoops. Had a different issue in mind. This looks like @ast just needs to rerun cmake (to rerun templating stuff) after pulling updates. This could plausibly go in as a feature request to add a target/intelligence to builds so that cmake gets rerun when new kernels are present

closed 2018-02-12 by n-west