gnuradio/volk #141: polarbutterfly: redundant argument

marcusmueller, 2018-01-29:
While trying to debug the issue of polarbutterflypuppet failing on AVX with a specific version of GCC only (not with clang), I figured out that the polarbutterfly kernel takes both

* `framesize` and
* `frame_exp`

with `framesize = (1<< frame_exp)` according to both documentation as well as puppet code. 

If this is really redundant, framesize should definitely be removed from the parameters. @jdemel, can you provide some input on why you have both?

marcusmueller, 2018-01-29:
## general comments

* Notice that the testcase failure *changes* when one replaces `frame_size` with `(1<<frame_exp)` in the avx kernel. This might mean we're onto something.
* I'm too tired right now, but `calculate_max_stage_depth_for_row` looks suspiciously much like it's actually a simple algebraic expression rather than something that needs to be solved in a loop. 
* `fabs` should be `fabsf` in `llr_odd`.
* The `return copysignf(1.0f, la) * copysignf(1.0f, lb) * (ala > alb ? alb : ala);` in `llr_odd` _does_ get some points for creativity!

marcusmueller, 2018-01-29:
@jdemel , what is the intention of `copysignf(1.0f, la) * copysignf(1.0f, lb) * (ala > alb ? alb : ala)`? What happens if either value (la, lb) is `0f` or `-0f`?

marcusmueller, 2018-01-29:
He's making a la/lb table,<br>
he's checkin' it twice.<br>
He's gonna find who's redundant or nice<br>
Minterm Santa's coming to town!

jdemel, 2018-01-30:
What's the exact output of the failing test case?
The expression `frame_size = (1<< frame_exp)` is indeed correct.
Also `u_num` and `row` are always called with the same value. This is a remainder of debugging this code.

This is basically the formula from the papers. `ala` and `alb` are absolute values.
`return copysignf(1.0f, la) * copysignf(1.0f, lb) * (ala > alb ? alb : ala);`

All in all, there are a lot of places for improvement. Let's discuss a feasable path forward.

noc0lour, 2018-02-01:
This is the output for builds with gcc (also happens with another distro using GCC 7.2):
https://ci.gnuradio.org/buildbot/#/builders/27/builds/1/steps/5/logs/stdio

marcusmueller, 2018-02-11:
To save us from later digging, here's copypasta from the CI Build log:

```
RUN_VOLK_TESTS: volk_32f_8u_polarbutterflypuppet_32f(131071,1)
generic completed in 30.301ms
u_avx completed in 3.957ms
offset 16 in1: -2 in2: 0 tolerance was: 1e-06
offset 17 in1: -4 in2: -2 tolerance was: 1e-06
offset 18 in1: -4 in2: 0 tolerance was: 1e-06
offset 19 in1: -8 in2: -4 tolerance was: 1e-06
offset 20 in1: -4 in2: 0 tolerance was: 1e-06
offset 21 in1: -8 in2: -4 tolerance was: 1e-06
offset 22 in1: -8 in2: -0 tolerance was: 1e-06
offset 23 in1: 16 in2: 8 tolerance was: 1e-06
offset 32 in1: -2 in2: 0 tolerance was: 1e-06
offset 33 in1: 4 in2: 0 tolerance was: 1e-06
volk_32f_8u_polarbutterflypuppet_32f: fail on arch u_avx
offset 33 in1: 0 in2: 1 tolerance was: 0
offset 34 in1: 0 in2: 1 tolerance was: 0
offset 36 in1: 0 in2: 1 tolerance was: 0
offset 37 in1: 0 in2: 1 tolerance was: 0
offset 38 in1: 0 in2: 1 tolerance was: 0
offset 39 in1: 0 in2: 1 tolerance was: 0
offset 41 in1: 0 in2: 1 tolerance was: 0
offset 42 in1: 0 in2: 1 tolerance was: 0
offset 44 in1: 0 in2: 1 tolerance was: 0
offset 45 in1: 0 in2: 1 tolerance was: 0
volk_32f_8u_polarbutterflypuppet_32f: fail on arch u_avx
Best aligned arch: generic
Best unaligned arch: generic
Failure on volk_32f_8u_polarbutterflypuppet_32f
```

noc0lour, 2018-02-11:
Also this affects at least GCC => 7.2 , but not GCC <= 5.2. I'm seeing this on my non-Fedora workstation with GCC 7.3 as well.

marcusmueller, 2018-02-11:
> All in all, there are a lot of places for improvement. Let's discuss a feasable path forward.

I think the easiest would be to ask the original author with very much "pretty please" to look over his code and remove the debugging facilities, and hence the redundant argument. With a bit of luck, we'd be seeing what might be going wrong there, if there is anything systematic. (Bonus: less cruft == probably more speed. And speed is what we are all here for, right?)

If this was but a single GCC version, I'd be convinced that maybe that specific version would be buggy; in this case, the good ol' "it's more likely my code than my compiler" rule of non-compiler-developer-thumb applies. @noc0lour, can you supply me with a minimal GCC version jump necessary to switch from working to non-working including platform/arch? I'd like to gcc.godbolt the hell outta this (maybe together with @jdemel, I guess we'd both learn something).

noc0lour, 2018-02-11:
Narf, compiling GCC is certainly not fun, but I'll try

noc0lour, 2018-02-11:
Updated my comment, apparently only GCC 7.{2,3}.0 is failing. Probably that's the case for the full GCC 7 series.

n-west, 2018-02-12:
I agree that this requires the original author to fix.

jdemel, 2018-02-12:
I investigated this issue. I could not reproduce the error yet. Need to get GCC7 in order to do so.

Issues so far are:
- use `frame_exp` only in kernels. I fixed that. BUT: Removing `frame_size` from the function call changes the API. How should we proceed here?
- `calculate_max_stage_depth_for_row` might be expressed more elegantly. I'm open to suggestions.
- `fabs` should be `fabsf` in `llr_odd`. Will do. Should be easy enough.
- `copysignf(1.0f, la) * copysignf(1.0f, lb) * (ala > alb ? alb : ala)` is actually the exact MinSum approximation formula. It says: sign(a) * sign(b) * min(abs(a), abs(b)). Again, I'm open to suggestions here.
- One more thing: I refactored quite a few places in the AVX kernel and ran into quite bizarre issues. Essentially, at the moment I have 2 `__mm256` variables that are initialized but unused. (Compiler complains about it too) But if I comment them out, the tests fail. And the reported error looks suspiciously like the one reported by the CI system. This might be a trace. Though, this needs further investigation.

noc0lour, 2018-02-12:
How about providing a function with `frame_size` which calls the implementation without `frame_size` and wrap that into a `DEPRECATED` macro? And then remove this function with the following release once it was in a release once?

marcusmueller, 2018-02-12:
Since I'm absolutely sure we're the only ones using that function yet, we don't need to wait two releases to fix one API. Fix it now. We never guarantee VOLK API to be stable across releases, do we, @nwest?

jdemel, 2018-02-21:
So the patches are merged. How does that affect the GCC 7 tests? Are they still present? Otherwise I'd suggest closing this issue.

n-west, 2018-02-22:
Going to close since it passes QA on travis (https://travis-ci.org/gnuradio/volk)

closed 2018-02-22 by n-west

noc0lour, 2018-02-22:
:+1: thanks :)

noc0lour, 2018-02-22:
@marcusmueller can you update the submodule for GNU Radio `next` to the tip of volk `master` again?

marcusmueller, 2018-02-22:
@noc0lour @n-west so be it.