gnuradio/volk #143: volk_32f_s32f_s32f_mod_range_32f.h doesn't compile with MSVC

ptrkrysik, 2018-02-01:
The errors can be seen here:
[05-Build GNURadio Release.txt](https://github.com/gnuradio/volk/files/1685321/05-Build.GNURadio.Release.txt)

MSVC compiler basically complains about lack of two intrinsics:
_mm_cvttps_pi32
_mm_cvtpi32x2_ps

This is funny as both of them are defined in xmmintrin.h that is included. In MSCV's xmmintrin.h they are inside of #if defined(_M_IX86) statement, but defining _M_IX86 manually didn't help.

guruofquality, 2018-02-02:
I think that anything MMX basically doesnt build for 64-bit MSVC. But since these are mmx intrinsics, it seems like they are missing the guard for LV_HAVE_MMX that would basically prevent this kernel from being built under this specific circumstance. So I think possible bug for LV_HAVEs on this kernel?

ptrkrysik, 2018-02-02:
These intrinsics are used inside of "#ifdef LV_HAVE_SSE". This is
probably why the compiler attempts to build this part.

ptrkrysik, 2018-02-02:
According to this pages these intrinsics are translated into SSE
instructions (not MMX ones):
https://msdn.microsoft.com/ru-ru/library/0d4dtzhb(v=vs.100).aspx
https://software.intel.com/en-us/node/524259

n-west, 2018-02-12:
if I read that correctly, you're using MSVC 2015 targeting 64-bit? Is there any relation between this and #134?

ptrkrysik, 2018-02-12:
Yes, I'm using MSVC 2015 to make a 64-bit build. I don't know how much
this issue is related to #134 as description of that issue lacks details
what was wrong.

On my side it seems the problem is only SSE part of
volk_32f_s32f_s32f_mod_range_32f kernel - with the two 'missing'
intrinsics that I pointed. These two intrinsics are defined xmmintrin.h
but somehow MSVC doesn't see them.

guruofquality, 2018-02-13:
I read this (https://msdn.microsoft.com/en-us/library/t467de55(v=vs.100).aspx)

> Any SSE intrinsics that use the __m64 data type are not supported on x64 processors.

So both _mm_cvttps_pi32 and _mm_cvtpi32x2_ps use the __m64 type. I thought we were guarding against this by using LV_HAVE_MMX in addition to LV_HAVE_SSE in this case. That lets us disable any kernel thats using __m64 when we dont have MMX support.

So thats my take on it (adding a `defined (LV_HAVE_SSE) && defined( LV_HAVE_MMX)`) if thats possible. I think this has come up before, dont remember. But the takeaway is we need to avoid building kernels on 64 bit when they use __m64 instructions.

@n-west I'm supposing that LV_HAVE_MMX should be undefined in this case, but is the define guard I suggested possible (to check for multiple arches like that)?

marcusmueller, 2018-02-13:
I think I've seen that problem elsewhere. It's strange, but possible, that unlike any other compiler, MSVC's `xmmintrin.h` just doesn't internally include `mmintrin.h`, and that somehow messes up the intrinsics. @ptrkrysik, can you try whether `#include <mmintrin.h>` inside the `#ifdef LV_HAVE_SSE`?

n-west, 2018-02-14:
I think I'm OK with `LV_GAVE_SSE && LV_HAVE_MMX`. The reality is that depending on MMX in the modern era probably means you're doing something wrong anyway. Can you make a PR?

ptrkrysik, 2018-02-14:
@marcusmueller, I tried to add the #include <mmintrin.h> inside of #ifdef LV_HAVE_SSE but it didin't help. In light of what @guruofquality has written it isn't surprising, as the _mm_cvttps_pi32 and _mm_cvtpi32x2_ps use the __m64 data type is not supported on x64.

ptrkrysik, 2018-02-14:
It seems to be specifically MSVC/Windows thing to not support __m64 on x64 architecture.

marcusmueller, 2018-02-14:
Writing down my observations: I think I found my fallacy; thanks everyone for bringing me very carefully to the observation that, indeed, SSE *does* have quadruple float->int32 and vice versa.

marcusmueller, 2018-02-14:
@ptrkrysik can you try my #153 branch, please?

ptrkrysik, 2018-02-15:
@marcusmueller I don't have good news yet. After trying to compile that branch I get different error message:

83>C:\gr-build\src-stage3\src\gnuradio\volk\kernels\volk/volk_32f_s32f_s32f_mod_range_32f.h(308): error C2679: binary '=': no operator found which takes a right-hand operand of type '__m128i' (or there is no acceptable conversion) (compiling source file C:\gr-build\src-stage3\build\Release\volk\lib\volk_machine_sse2.c) [C:\gr-build\src-stage3\build\Release\volk\lib\volk_obj.vcxproj]
         C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\include\xmmintrin.h(75): note: could be '__m128 &__m128::operator =(__m128 &&)' (compiling source file C:\gr-build\src-stage3\build\Release\volk\lib\volk_machine_sse2.c)
         C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\include\xmmintrin.h(75): note: or       '__m128 &__m128::operator =(const __m128 &)' (compiling source file C:\gr-build\src-stage3\build\Release\volk\lib\volk_machine_sse2.c)
         C:\gr-build\src-stage3\src\gnuradio\volk\kernels\volk/volk_32f_s32f_s32f_mod_range_32f.h(308): note: while trying to match the argument list '(__m128, __m128i)' (compiling source file C:\gr-build\src-stage3\build\Release\volk\lib\volk_machine_sse2.c)
    83>C:\gr-build\src-stage3\src\gnuradio\volk\kernels\volk/volk_32f_s32f_s32f_mod_range_32f.h(309): error C2664: '__m128 _mm_cvtepi32_ps(__m128i)': cannot convert argument 1 from '__m128' to '__m128i' (compiling source file C:\gr-build\src-stage3\build\Release\volk\lib\volk_machine_sse2.c) [C:\gr-build\src-stage3\build\Release\volk\lib\volk_obj.vcxproj]
         C:\gr-build\src-stage3\src\gnuradio\volk\kernels\volk/volk_32f_s32f_s32f_mod_range_32f.h(309): note: No user-defined-conversion operator available that can perform this conversion, or the operator cannot be called (compiling source file C:\gr-build\src-stage3\build\Release\volk\lib\volk_machine_sse2.c)
    83>C:\gr-build\src-stage3\src\gnuradio\volk\kernels\volk/volk_32f_s32f_s32f_mod_range_32f.h(364): error C2679: binary '=': no operator found which takes a right-hand operand of type '__m128i' (or there is no acceptable conversion) (compiling source file C:\gr-build\src-stage3\build\Release\volk\lib\volk_machine_sse2.c) [C:\gr-build\src-stage3\build\Release\volk\lib\volk_obj.vcxproj]
         C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\include\xmmintrin.h(75): note: could be '__m128 &__m128::operator =(__m128 &&)' (compiling source file C:\gr-build\src-stage3\build\Release\volk\lib\volk_machine_sse2.c)
         C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\include\xmmintrin.h(75): note: or       '__m128 &__m128::operator =(const __m128 &)' (compiling source file C:\gr-build\src-stage3\build\Release\volk\lib\volk_machine_sse2.c)
         C:\gr-build\src-stage3\src\gnuradio\volk\kernels\volk/volk_32f_s32f_s32f_mod_range_32f.h(364): note: while trying to match the argument list '(__m128, __m128i)' (compiling source file C:\gr-build\src-stage3\build\Release\volk\lib\volk_machine_sse2.c)
    83>C:\gr-build\src-stage3\src\gnuradio\volk\kernels\volk/volk_32f_s32f_s32f_mod_range_32f.h(365): error C2664: '__m128 _mm_cvtepi32_ps(__m128i)': cannot convert argument 1 from '__m128' to '__m128i' (compiling source file C:\gr-build\src-stage3\build\Release\volk\lib\volk_machine_sse2.c) [C:\gr-build\src-stage3\build\Release\volk\lib\volk_obj.vcxproj]
         C:\gr-build\src-stage3\src\gnuradio\volk\kernels\volk/volk_32f_s32f_s32f_mod_range_32f.h(365): note: No user-defined-conversion operator available that can perform this conversion, or the operator cannot be called (compiling source file C:\gr-build\src-stage3\build\Release\volk\lib\volk_machine_sse2.c)
   362>CustomBuild:

The full build log is here:
[05-Build GNURadio Release.txt](https://github.com/gnuradio/volk/files/1726941/05-Build.GNURadio.Release.txt)

marcusmueller, 2018-02-16:
Ok, interesting that my compiler didn't slap me in the face for that. I've fixed those, and don't get warnings, but lacking a Windows machine right now, I kind of depend on you to test that fix... :(

ptrkrysik, 2018-02-16:
@marcusmueller, I tried to compile it but it failed again.  So I decided to try fix something myself to make the code at least compile and then give it to you to check what I have done. The result is here:
https://github.com/ptrkrysik/volk/tree/modrange_fix_mmx

I noticed that 'rounddown' variable is abandoned after computing it so I tried to do something with it - I just gave it to the input of the next intrinsic that failed to compile. The kernel got compiled after it, but keep in mind that at this moment I don't understand what this code is trying to achieve and this fix was just random guess.

There was also unrelated problem with volk_profile.cc - 'not' logical statement that didn't compile with MSVC. I replaced it with '!'.

marcusmueller, 2018-02-16:
oh yeah! Your guess was totally on-spot. I'll cherry-pick your amendment to the kernel onto my branch and make that a fixup commit (to be squashed away later).

Regarding the volk_profile fix: let's do that on a different PR.

PR #153 (2018-02-20): Replaced 2× 2float->2int32 to 1× 4float->4int32
As the dual operand operations were MMX, and that was causing trouble on
MSVC on x86_64.

I wonder why I didn't do that in the first place; it should be
significantly faster.

Hopefully fixes #143.
commit 65ced911 (2018-02-20): Replaced 2× 2float->2int32 to 1× 4float->4int32

As the dual operand operations were MMX, and that was causing trouble on
MSVC on x86_64.

I wonder why I didn't do that in the first place; it should be
significantly faster.

Hopefully fixes #143.
closed 2018-02-20 by n-west