gnuradio/volk #151: Proposal: New kernel for mixing

Ka-zam, 2018-02-13:
Often in SDR workloads you find yourself needing to mix two signals:

y(t) = x(t)*alpha*exp(j*w*t) = alpha*( Re(x) + Im(x) )*( cos(w*t) + j*sin(w*t) )

This can be accomplished with VOLK using sin/cos, multiply and interleave. 
For efficiency I propose a new kernel:

volk_32fc_32f__s32f_mix_32fc (lv_32fc_t* cVector, const lv_32fc_t* bVector, const float* aVector, const float alpha, unsigned int num_points)

The main efficiency would come from calculating sin/cos at the same time. 

Alternatively a complex valued exponential function and/or a combined sin/cos function would be nice.

Feasible?

rear1019, 2018-02-13:
See `volk_32fc_s32fc_x2_rotator_32fc` [1].

[1] http://libvolk.org/doxygen/volk_32fc_s32fc_x2_rotator_32fc.html

n-west, 2018-02-14:
As pointed at, this kernel exists.

closed 2018-02-14 by n-west