gnuradio/volk #161: using travis-ci

hcab14, 2018-02-21:
Hi @n-west, 

you mentioned here https://github.com/gnuradio/volk/pull/145#issuecomment-363600544, that you do not have an ARM setup for testing. I realized that one can use travis-ci for this, see, e.g., this log file
https://travis-ci.org/hcab14/volk/builds/344523682
and my fork https://github.com/hcab14/volk

boost and volk are build using `arm-linux-gnueabihf-gcc` and `volk_test_all` is executed using `qemu-arm`.

Let me know what you think. After adding compilation with standard GCC I can open a pull request. It would be easy to add builds with several versions of GCC, too.

noc0lour, 2018-02-21:
I think @n-west already has some almost done travis setup here: https://github.com/n-west/volk 
Probably you could pick that up and add ARM building to it?

marcusmueller, 2018-02-21:
Please note that this is not a discussion forum, but a bug tracker!

That contribution is very welcome, but I'd recommend discussing this on the discuss-gnuradio mailing list. Thank you!

noc0lour, 2018-02-21:
@marcusmueller I'd say this is rather on point for this bug tracker even if it is not a bug in the code but in the deployment. And the author already expressed intend to fix it.

marcusmueller, 2018-02-21:
I'd happily accept the perspective of these involved directly, so:

Carry on as if I said nothing!

hcab14, 2018-02-22:
Hi @marcusmueller, all,

> Please note that this is not a discussion forum, but a bug tracker!
> That contribution is very welcome, but I'd recommend discussing this on the discuss-gnuradio mailing list. Thank you!

I see your point.

In another repository I got used to discuss enhancements in GitHub issues, and there labels are used to distinguish between bugs, proposals for enhancements, and other kind of issues.

The discussion can now move to this pull request https://github.com/gnuradio/volk/pull/162

closed 2018-02-22 by hcab14

n-west, 2018-02-22:
I think it's OK to have this discussion here. It'll get my attention better than the mailing list and is directly related to code changes.

Yea, I have a PoC travis file on my personal fork. I squashed all of the commits in to one last night when it was linked-- I can merge that in and fix the outstanding c99 issues. Do you have something that already works with qemu? Is it better to merge that in first and you can use it or use whatever you might have? My interest in travis was

 * verifying older compiler support is kept
 * verifying builds and QA on "not my machine"

if it can also test different architectures that would be a major win as well. I'll test out that travis-- do you have a public URL with results already?