gnuradio/volk #168: volk_32fc_x2_dot_prod_32fc sse 4.1 fail on Win64/MSVC 2015

gnieboer, 2018-03-21:
Output from test:

RUN_VOLK_TESTS: volk_32fc_x2_dot_prod_32fc(131071,1)
generic completed in 0ms
u_sse3 completed in 0ms
u_sse4_1 completed in 1ms
u_avx completed in 0ms
a_generic completed in 0ms
a_sse3 completed in 0ms
a_sse4_1 completed in 1ms
a_avx completed in 0ms
offset 0 in1: 36.7184 + -68.6857j  in2: 154.464 + -68.6862j tolerance was: 0.01
volk_32fc_x2_dot_prod_32fc: fail on arch u_sse4_1
offset 0 in1: 36.7184 + -68.6857j  in2: 154.464 + -68.6862j tolerance was: 0.01
volk_32fc_x2_dot_prod_32fc: fail on arch a_sse4_1
Best aligned arch: generic
Best unaligned arch: generic

jdemel, 2019-11-04:
@gnieboer could you provide more info on that issue? e.g. does it fail reliably if you only run this test?

michaelld, 2019-11-14:
@gnieboer is this issue still valid with the current Volk GIT master? We've fixed a bunch of kernel issues in the recent while!

jdemel, 2020-01-25:
cf. #170 We didn't hear from @gnieboer  for a while. I just close it now. If the problem is still present, just reopen it.

closed 2020-01-25 by jdemel