gnuradio/volk #170: volk_32f_x2_pow_32f fails SSE 4.1 test on Win64/MSVC 2015

gnieboer, 2018-03-21:
RUN_VOLK_TESTS: volk_32f_x2_pow_32f(131071,1)
a_sse4_1 completed in 1ms
generic completed in 3ms
u_sse4_1 completed in 0ms
offset 25199 in1: -4.13255 in2: 4.13197 tolerance was: 0.01
offset 70001 in1: -0.169713 in2: 0.169816 tolerance was: 0.01
volk_32f_x2_pow_32f: fail on arch a_sse4_1
offset 25199 in1: -4.13255 in2: 4.13197 tolerance was: 0.01
offset 70001 in1: -0.169713 in2: 0.169816 tolerance was: 0.01
volk_32f_x2_pow_32f: fail on arch u_sse4_1
Best aligned arch: u_sse4_1
Best unaligned arch: u_sse4_1

michaelld, 2019-11-14:
@gnieboer is this issue still valid with the current Volk GIT master? We've fixed a bunch of kernel issues in the recent while!

jdemel, 2020-01-25:
Since we didn't hear back from @gnieboer , I just close this issue. If this is still present please just reopen.

closed 2020-01-25 by jdemel