gnuradio/volk #171: Need a VOLK release ASAP.

marcusmueller, 2018-03-22:
In order to release GR 3.7.12, I'd need a VOLK release this week.

n-west, 2018-03-24:
v1.3.1 out in tag

marcusmueller, 2018-03-24:
Awesome! Thank you!

marcusmueller, 2018-03-24:
Awesome! Thank you!

marcusmueller, 2018-03-26:
Nope, sadly have to reopen this: I need #155 in that release :( sorry

marcusmueller, 2018-03-26:
There's a branch called [apply_PR_#155](https://github.com/gnuradio/volk/tree/apply_PR_%23155); please merge onto maint, and tag appropriately!

marcusmueller, 2018-03-26:
old development model. Last release to do so.

n-west, 2018-03-26:
I've been doing the merge back model, actually. @hcab14 maint is sort of a longer term support branch because some people really need a stable API more than they need new features. To accommodate that I make everything go on master, cherry-pick bug fixes back to maint. In theory maint is released every so often and people can safely update with confidence of getting bug fixes and not broken code/dependencies. In practice I don't have as strict a release schedule as I should and have recently been lazy about releases

n-west, 2018-03-27:
Released v1.4!

closed 2018-03-27 by n-west