gnuradio/volk #174: purge boost dependency

jdemel, 2018-03-27:
I'd like to open this issue as a reference. In the 1.3.1/1.4 release announcement was a remark to purge boost from VOLK. I consider this issue a place to discuss the required changes.
A quick `grep -irnI boost .` reveals essentially to files which still contain `boost`.

Affected files:
- `lib/qa_utils.h`
`BOOST_AUTO_TEST_CASE` and `BOOST_CHECK_EQUAL`. Since there is no `#include <boost>`, I assume this is just a naming thing. Though, there is also some CMake magic in `lib/CMakeLists.txt`. I suggest to rename these macros and remove the boost reference in the corresponding CMake file.
- `apps/volk_profile.cc`
`boost::filesystem`. This is probably the tough one. C++11 does not support any filesystem magic. All I could find is a pointer to some experimental C++17 features. I guess C++17 is  way to new for VOLK right now. Boost is only used in the function `write_results` with `boost::filesystem::path`, `boost::filesystem::exists` and `boost::filesystem::create_directories`. I suggest to purge the defined namespace `fs` in order to make it more obvious that Boost is used here. And then we need to find an elegant way to remove these function calls. Maybe someone can come up with a good solution here.

noc0lour, 2018-03-27:
I'd vote for a single header file (or a implementation written by someone else) instead of including Boost. It's really just two lines for creating the configuration directory structure. That's not a reason to still have boost around. Maybe we can have something which wraps a simple implementation and `<filesystem>` or backport `<filesystem>` to volk.

marcusmueller, 2018-03-27:
@noc0lour 

* https://github.com/ZhreShold/zupply/wiki/Filesystem

n-west, 2018-03-27:
Thanks @jdemel.

Removing boost is 100% worth it because it’s a huge PITA to link against and there are non-gnuradio uses of VOLK including mine which would greatly benefit from dropping boost as a dependency. And yes, the last bit of (used) boost is in creating the config directory.

I would like either conditionalized c++17 support or a simple header only solution that can be dropped in. I haven’t looked at merits of options yet. The reason this is the last bit left is because I didn’t want to deal with the nightmare of cross-platform directory trees myself.

noc0lour, 2018-03-27:
@marcusmueller The source code doesn't look to good, I've looked up something else yesterday, but it was missing the tree creation feature. 

Also thought of ripping out the `create_directories()` from Boost. This is not really viable either.

marcusmueller, 2018-03-27:
@noc0lour bummer for you: I've been looking for exactly this functionality before, and there's very few "slim" libraries that do it, because you get all the niceness of dealing with different concepts between different OSes, starting with whether a directory is like a file, how you recursively create directories (and tokenize a path) on  things that are as similar as Windows and Linux, the goodness of having to deal with different OS APIs, which all fail very differently. Oh, and of course, things like the OS doing file name UTF normalization beneath the hood, so that the directory you just created simply can't be accessed with the same byte string name you used to create it (thanks, OS X!).

So, if you were expecting beautiful AND short code, how about you roll a bit in GCC 2.6? `</sarcasm>`

I don't think a portable, nice-to-read lib exists that's much less of a pain than Boost. Poco comes to mind, but that's just all the shortcomings of Boost in a different packaging.

n-west, 2018-03-27:
much experimental. https://github.com/n-west/volk/commit/bfd4f7858efd15c0f20abbeddc2f763f31caf7b9

(https://stackoverflow.com/questions/48729328/cant-use-any-filesystem-members-in-c17-gcc-7)

noc0lour, 2018-03-27:
We could just test if the directory is created and just fail/not write config if it's not created?

hcab14, 2018-03-27:
What do you think about wrapping `volk_profile` in a shell script (linux+osx) or a .bat file (windows) which creates the directory if needed, and then launches `volk_profile` from there?

jdemel, 2018-03-28:
VOLK depends on Python anyway, so it might be a solution to run the `volk_profile` interface as a Python script. Though, Nathan's solution is less invasive.

noc0lour, 2018-03-28:
though it would bump python from build dependency to runtime dependency.

alexxcons, 2019-01-11:
+1, would be great to purge that boost dependeny !

Currently I have massive trouble on a system with preinstalled gnuradio at work, since I have to link with some other package which uses a boost version which is newer than the one provided by the system.

alexxcons, 2019-01-15:
Thanks for the hint! Yes I will need to build GR from source against the newer boost version and install it in my home directory. ( Dont have root on the system )

jdemel, 2019-05-24:
PR #251 will help to purge Boost. Also, it should make it easier to compile VOLK without Boost.

jdemel, 2019-09-22:
PR #255 is a better solution to purge the boost dependency.

It enables users to build VOLK without Boost on modern systems because it detects the necessary `std::filesystem` features automatically. Still, users can build VOLK with Boost on older systems. At some point in the future we can just drop this check and thus finally purge boost completely.

jdemel, 2019-12-07:
Since #255 got merged, we are on track to purge boost. At some point in the future, we can drop boost as a dependency. Though, this will take time because we need to make sure that all supported compilers support `std::filesystem` at that point. 

I leave this issue open to remind us of our goal to purge boost.

jdemel, 2021-04-11:
With #463 the last CI test that requires Boost is removed. It seems like it is about time to purge boost.

michaelld, 2021-04-12:
MacOS will still require Boost for filesystem ... either that or require GCC for the compiler because Clang's std:: filesystem implementation isn't complete even for current Clang (to the best of my memory / knowledge / testing). Given that we're trying to still support current macOS (10.15 as well as 11.X), we must require Boost for filesystem for those, but for macOS only. I think we have the logic already setup to do this. All other Boost can be purged / switched to std:: if not done already.

jdemel, 2021-04-12:
I'm quite surprised. Our GH CI seems to work without Boost. But it could be part of the base image. Whenever we can detect `std::filesystem` or `std::experimental::filesystem`, we use it instead of Boost. That's the only remaining Boost use case. Of course, I'd keep Boost around as long as necessary.

PR #557 (2022-02-07): boost: Remove boost dependency
VOLK used Boost in the past quite heavily. We reduced Boost usage since then. At the moment, we only require `boost::filesystem` for very old systems. For all recent systems we rely on `std::filesystem` or `std::experimental::filesystem`.

The latest Ubuntu version that still required Boost was 14.04. Starting with Ubuntu 16.04, we can use `std::experimental::filesystem` and later `std::filesystem`.

Fix #174
commit 53251404 (2022-02-07): boost: Remove boost dependency

VOLK used Boost in the past quite heavily. We reduced Boost usage since
then. At the moment, we only require `boost::filesystem` for very old
systems. For all recent systems we rely on `std::filesystem` or
`std::experimental::filesystem`.

The latest Ubuntu version that still required Boost was 14.04. Starting
with Ubuntu 16.04, we can use `std::experimental::filesystem` and later
`std::filesystem`.

Fix #174
closed 2022-02-07 by michaelld