gnuradio/volk #175: VOLK always assumes neon in arm builds. Also cmake needs updating to build for 32 and 64 bit arm

balister, 2018-04-02:
This is not always true.

For 32 bits build, cmake needs to inspect the flags passed to the compiler and see if -mfpu=neon is set. (Or do a compile time test), if this is true it should build and 32 bit neon assembly and arm intrinsics code.

If it is an aarch64 machine, it can assume 64 bit neon instructions. So build 64 bit asm and arm intrinsics.

32 and 64 bit neon asm is different, but the intrinsics work for both.

michaelld, 2019-11-14:
@balister Is this still an issue? If so, what's a good approach to deal with it?

jdemel, 2020-01-14:
Since we check for NEON presence in #319 and then for the correct version, I'd assume this issue is solved.
I propose to close it unless @balister says otherwise.

michaelld, 2020-01-24:
Closing since there are no comments from the @balister gallery. Please reopen if this is still an issue, and explain what's going on so that we can address it!

closed 2020-01-24 by michaelld