gnuradio/volk #177: Strange fienames in tarball

wkazubski, 2018-04-05:
Names of files and directores inside the tarball of volk 1.4 published on http://libvolk.org/releases.html webpage are strange. Probably a slash was omited while packaging. Please correct.

n-west, 2018-04-05:
Yea, it's just got everything prefixed with volk-1.4 by accident. I'll fix it when I get some time on the machine that can also sign the tar balls. Thanks for the report

stef, 2018-07-31:
i just wanted to report the same, still unfixed it seems.

michaelld, 2019-11-04:
Is this still an issue with Volk 2.0.0 release?

jdemel, 2019-11-05:
@n-west needs to update [libvolk.org](http://libvolk.org/releases.html) to fix this issue. Release 2.0 is not yet on the webpage.
I guess we need to figure out how we're going to handle the website.

michaelld, 2019-11-14:
Do we know who owns libvolk.org & how do we gain access to it to make changes? Seems like the ability to make changes should go with Volk maintainers, somehow...

jdemel, 2020-02-08:
We sorted out how we handle the website. Thus, we'll be able to provide tarballs for new releases soon. There are already tarballs for 2.0 and 2.1 on the website. And soon we'll have a signed tarball for 2.1 as well.

The new tarballs should not have the 'strange tarball' issue. I suppose we can close this issue as soon as we have those new tarballs up. I assume it won't be possible to fix old release tarballs though. I hope that's OK.

jdemel, 2020-02-22:
By now, VOLK 2.0 to 2.2 tarballs are on the website. All of them should be correct in terms of filenames. Thus, I'd consider this issue fixed.

I suggest we can close this issue now. @michaelld

wkazubski, 2020-02-22:
Closing the issue

closed 2020-02-22 by wkazubski