gnuradio/volk #186: INT_MIN is wrong

soggysec, 2018-05-22:
MAX_INT =  (2^31)-1 // 2147483647
MIN_INT =  (-MAX_INT - 1)  // -2147483648

// Should replace all uses of the wrong value with -2147483648, OR better yet:

```
std::numeric_limits<int>::max()
std::numeric_limits<int>::min()
```

Uses of wrong value:
```
grep -rn "\-2147483647" . 
./volk/kernels/volk/volk_32f_s32f_convert_32i.h:90:  float min_val = -2147483647;
./volk/kernels/volk/volk_32f_s32f_convert_32i.h:138:  float min_val = -2147483647;
./volk/kernels/volk/volk_32f_s32f_convert_32i.h:185:  float min_val = -2147483647;
./volk/kernels/volk/volk_32f_s32f_convert_32i.h:225:  float min_val = -2147483647;
./volk/kernels/volk/volk_32f_s32f_convert_32i.h:273:  float min_val = -2147483647;
./volk/kernels/volk/volk_32f_s32f_convert_32i.h:321:  float min_val = -2147483647;
./volk/kernels/volk/volk_32f_s32f_convert_32i.h:368:  float min_val = -2147483647;
```

n-west, 2018-05-22:
Can you rationalize your desired change with http://www.cplusplus.com/reference/climits/? Sure, I'm happy to replace a hard definition with a compiler defined const but it has to be from limits.h since that file is a C file and not C++ (no std::).

soggysec, 2018-05-22:
Funny Enough, cplusplus.com is actually wrong as well. Below is a sample program using INT_MIN and limits.h

```
cat /tmp/test.c 
#include <limits.h>

int main(){
    printf("%d\n",INT_MIN);
    return 0;
}
user:~/git/gnuradio $ /tmp/test 
-2147483648
```

soggysec, 2018-05-22:
also from /usr/include/limits.h
```
/* Minimum and maximum values a `signed int' can hold.  */
 #  define INT_MIN   (-INT_MAX - 1)
 #  define INT_MAX   2147483647
```

jdemel, 2019-11-04:
[The CPPreference docs](https://en.cppreference.com/w/c/types/limits) seem to point to 
```
INT_MIN    = -2147483648
INT_MAX    = +2147483647
```
in `limits.h`. I assume we should do add `#include <limits.h>` and remove our custom definitions. Is there a reason why we have custom definitions? e.g. compiler issues?

michaelld, 2019-11-04:
:+1: from me to move to using `<limits.h>` instead of hard-coding. I see no good reason why not to! Let's try & see what happens ...

jdemel, 2019-11-06:
Since #298 got merged, I suggest to close this issue.

michaelld, 2019-11-06:
Agreed!

closed 2019-11-06 by michaelld