gnuradio/volk #187: implicit declaration of `posix_memalign`: missing include in `volk_malloc.c`?

marcusmueller, 2018-06-04:
<pre>
[97/701] Building C object volk/lib/CMakeFiles/volk_obj.dir/volk_malloc.c.o
../volk/lib/volk_malloc.c:49:13: warning: implicit declaration of function 'posix_memalign' is invalid in C99 [-Wimplicit-function-declaration]
  int err = posix_memalign(&ptr, alignment, size);
            ^
1 warning generated.
</pre>

jdemel, 2019-11-07:
I couldn't reproduce this warning. I guess we need some more input on this. How did this happen? Is this still an issue? If not, I'd prefer to close this issue.

michaelld, 2019-11-08:
We `#include` the obvious headers where `posix_memalign` is located on most OSs. It's always possible there's some OS where this function is found in some non-standard header ... without more info there's really not anything we can do!

michaelld, 2019-11-08:
I'm closing this issue. If new info comes up on the OS / version to test / verify, then please post it & reopen this issue.

closed 2019-11-08 by michaelld