gnuradio/volk #188: `32f_s32f_convert_8i`: off-by-one errors

marcusmueller, 2018-06-09:
Paul Boven writes:

> Hi everyone,
> I
[image]
'm trying to perform 2 bit sampling of an RF signal. In one approach, 
> I'm using a float->char block, and noticed that around zero, a number of 
> float inputs become quantized in a bin that's one off from the correct 
> value. The ones that are wrong are always off by one, with their 
> quantization error always in the direction of zero.
> 
> The problem can be demonstrated with a really simple flowchart, using 
> the following blocks:
> 
> * Noise Source (Noise Type: Gaussian, Amplitude: 1, Seed: 0, Output 
> type: Float)
> * Throttle
> The throttle is then connected to two blocks:
> * A file-sink (Type Float) and a
> * 'Float to Char' block.
> * The float to char block is again connected to a File Sink, now of type 
> Char.
> 
> As an example, I've plotted all the samples that quantized as zero. 
> These should fall in the range [-0.5:0.5], but occasionally we also get 
> hits that lie within [-1:1]. These mishaps are rare (about one in 2000). 
> It also shows that they only occur at multiples of 8192 samples, and 
> zooming in reveals that they always happen shortly before the next 
> multiple of 8192, not after.
> 
> For other values than 0, the same applies, but the misquantizations are 
> only in one direction, ending up in a lower bin if the input is 
> positive, or in a higher bin if the input is negative. Again, the 
> misquantizations only occur in half the bin: For a value of 1, the float 
> value should be in [0.5:1.5], but occasionally a value in [1.5:2] also 
> ends up being quantized as 1.
> 
> This seems to me a bug that is somehow related to internal block 
> boundaries, but I'm not familiar enough with the internals of GnuRadio 
> to figure out just what's going wrong.
> 
> The problem does NOT occur when converting to Short or Int.
> 
> This is using GnuRadio 3.7.11 (as packaged with Ubuntu 18.04).
> 
> Regards, Paul Boven.

marcusmueller, 2018-06-09:
Can reproduce: when setting the tolerance for the convert kernel to 0 in `kernel_tests.h`, we get multiple failures of ONLY the SSE2 kernel, both aligned and unaligned.

```
44: Test command: /usr/bin/sh "/home/marcus/src/gnuradio/volk/build/lib/volk_32f_s32f_convert_8i_test.sh" "/home/marcus/src/gnuradio/volk/build/lib"
44: Test timeout computed to be: 10000000
44: RUN_VOLK_TESTS: volk_32f_s32f_convert_8i(131071,1)
44: u_sse2 completed in 0.048ms
44: u_sse completed in 0.08ms
44: generic completed in 0.08ms
44: a_sse2 completed in 0.041ms
44: a_sse completed in 0.08ms
44: a_generic completed in 0.062ms
44: offset 6 in1: -107 in2: -108 tolerance was: 0
44: offset 10 in1: -14 in2: -15 tolerance was: 0
44: offset 13 in1: 8 in2: 9 tolerance was: 0
44: offset 16 in1: 88 in2: 89 tolerance was: 0
44: offset 19 in1: 69 in2: 70 tolerance was: 0
44: offset 25 in1: -64 in2: -65 tolerance was: 0
44: offset 32 in1: 73 in2: 74 tolerance was: 0
44: offset 34 in1: 89 in2: 90 tolerance was: 0
44: offset 35 in1: 15 in2: 16 tolerance was: 0
44: offset 53 in1: 106 in2: 107 tolerance was: 0
44: volk_32f_s32f_convert_8i: fail on arch u_sse2
44: offset 6 in1: -107 in2: -108 tolerance was: 0
44: offset 10 in1: -14 in2: -15 tolerance was: 0
44: offset 13 in1: 8 in2: 9 tolerance was: 0
44: offset 16 in1: 88 in2: 89 tolerance was: 0
44: offset 19 in1: 69 in2: 70 tolerance was: 0
44: offset 25 in1: -64 in2: -65 tolerance was: 0
44: offset 32 in1: 73 in2: 74 tolerance was: 0
44: offset 34 in1: 89 in2: 90 tolerance was: 0
44: offset 35 in1: 15 in2: 16 tolerance was: 0
44: offset 53 in1: 106 in2: 107 tolerance was: 0
44: volk_32f_s32f_convert_8i: fail on arch a_sse2
44: Best aligned arch: a_sse2
44: Best unaligned arch: u_sse2
1/1 Test #44: qa_volk_32f_s32f_convert_8i ......***Failed    0.01 sec

```

marcusmueller, 2018-06-09:
The SSE2 and above kernels do something fundamentally different from what the generic kernel does: 

While the generic kernel simply does a `out = (int8_t) scaled_in`, the SIMD kernels do a `out = saturated_conversion<int8_t>(saturated_conversion<int16_t>(scaled_in))`. This means that generic maps `128.0f` to `0`, whereas SSE2 and above map `128.0f` to `127`.

Which conversion do we want?

jdemel, 2019-11-04:
Is this still an issue? We had multiple PRs that tried to fix this issue and related ones. I'd assume this issue can be closed.

hcab14, 2019-11-04:
AFAIC this issue can be closed.

Testing with the following command
```sh
while [ 1 ]; do volk_profile -n -i 1 -R 32f_s32f_convert_8i; done | grep off
```
does not output anything.

michaelld, 2019-11-04:
Thanks for the point! Closing then. If this remains an issue for anyone, please reopen and provide debugging info for what's going on.

closed 2019-11-04 by michaelld