gnuradio/volk #191: Errors while running CTest

ericwang321, 2018-07-22:
================================================
> pyenv version
_2.7.12 (set by /root/.pyenv/version)_
> uname -a
_Linux unix 4.4.0-130-generic #156-Ubuntu SMP  i686 i686 i686 GNU/Linux_
==================================================
install volk  as  instruction  from website.  when run "make test",output appears
```
Total Test time (real) =   4.41 sec

The following tests FAILED:
	  7 - qa_volk_32f_x2_fm_detectpuppet_32f (Failed)
	 28 - qa_volk_32f_s32f_mod_rangepuppet_32f (Failed)
Errors while running CTest
Makefile:94: recipe for target 'test' failed
make: *** [test] Error 8
```
=======================================
  how to solve this problem? google search return is nothing .
  Thanks.

michaelld, 2018-07-22:
For further elaboration purposes to help debug, what do the following return:
{{{
volk-config-info -v
volk-config-info --cc
ctest -V -R qa_volk_32f_x2_fm_detectpuppet_32f
ctest -V -R qa_volk_32f_s32f_mod_rangepuppet_32f
}}}

ericwang321, 2018-07-24:
thanks.
but I have  deleted many installed software directory . os is choas.
the error message cannot appear. it appear other problem.
so  i have to reinstall the OS ......

closed 2018-07-24 by ericwang321