gnuradio/volk #193: Document tolerance for a passed null pointer in a specific function parameter

elfring, 2018-07-31:
[The function “volk_free”](https://github.com/gnuradio/volk/blob/17277ed3130e9b2592da06c681657e63b5590c56/lib/volk_malloc.c#L61 "Implementation of volk_free()") could be implemented by a simple call of [the function “free”](http://pubs.opengroup.org/onlinepubs/9699919799/functions/free.html "Free allocated memory").
Thus it seems that this function implementation inherits the property to tolerate a null pointer which was received in the parameter “aptr”.

:bulb: How do you think about to express this detail also in [the corresponding programming interface description](https://github.com/gnuradio/volk/blob/ddd9c348440f11317619da2802c369a8e980bef9/include/volk/volk_malloc.h#L59 "Update candidate in the API description")?

jdemel, 2019-11-07:
Since `volk_free` just calls `free`, I'd say this is a good idea. Would you be willing to issue a PR for this @elfring ?

michaelld, 2019-11-14:
I like this idea too. Maybe someone can do a PR for it?

jdemel, 2020-01-23:
I had an issue with `volk_free` recently. Basically, `volk_free(NULL)` or `volk_free(nullptr)` throw a segfault. This is not the case on all machines. I've used such code for quite a while on Intel machines but on a new AMD system this throws a segfault. All systems run Ubuntu 18.04. 

```
#include <volk/volk.h>
#include <cstdio>

int main(int argc, char **argv) {
  std::complex<float>* vec = nullptr;
  unsigned N = 100;
  unsigned int alignment = volk_get_alignment();
  lv_32fc_t* in  = (lv_32fc_t*)volk_malloc(sizeof(lv_32fc_t)*N, alignment);
  lv_32fc_t* out = (lv_32fc_t*)volk_malloc(sizeof(lv_32fc_t)*N, alignment);
  
  volk_free(in);
  volk_free(out);
  free(NULL);
  volk_free(NULL);
  //volk_free(vec);

}
```
Compile with:
`g++ -Wall -g -I[PREFIX]/include -L[PREFIX]/lib volk_free_test.c -lvolk -lm`

We have different implementations for `volk_free`. Not all of them just call `free`.
1. `free`
2. `_aligned_free`
3. see below
```
void
volk_free(void *ptr)
{
  struct block_info *info;

  /* Get the real pointer */
  info = (struct block_info *)(((uintptr_t)ptr) - sizeof(struct block_info));

  /* Release real pointer */
  free(info->real);
}
```
That's the one that fails for me. I assume some preprocessor defines do not work as expected.

I assume [aligned_alloc](https://en.cppreference.com/w/c/memory/aligned_alloc) is the prefered function for us. Which can be used in conjunction with `free` on all systems.

Since we require C11, we can expect `aligned_alloc`. I suggest we drop all other implementations and just use this one.

michaelld, 2020-01-24:
Agreed! Looks totally legit to move to more recent C functions that are provided by C11! Good catch!

PR #334 (2020-02-15): alloc: Use C11 aligned_alloc
We require C11. Thus, we can now rely on `aligned_alloc` and `free`.
[aligned_alloc](https://en.cppreference.com/w/c/memory/aligned_alloc)
Also, the documentation is updated accordingly.

Closes #193
closed 2020-02-15 by michaelld