gnuradio/volk #194: reserved identifier violation

elfring, 2018-07-31:
I would like to point out that an identifier like “[`__VOLK_ASM`](https://github.com/gnuradio/volk/blob/a0e406e042ed9cda27333efa2de0fcc10912ea07/include/volk/volk_common.h#L12 "Update candidate")” [does eventually not fit](https://www.securecoding.cert.org/confluence/display/cplusplus/DCL51-CPP.+Do+not+declare+or+define+a+reserved+identifier "Do not declare an identifier which is reserved for the compiler implementation.") to the expected naming convention of the C++ language standard.
Would you like to adjust your selection for unique names?

jdemel, 2019-11-09:
I assume C++ and C naming conventions are aligned on this one. According to the naming convention page you linked, we should drop the leading `__`. We need to discuss why leading underscores were introduced in the first place. Also we need to discuss if it is save to remove them and if we want to remove them.

michaelld, 2019-11-14:
Fascinating. I had no idea there were so many rules for naming conventions in C++. Crazy!

I have no objections to renaming any macro / identifier / function / variable, so long as the code still builds on a variety of OSs and using multiple different compilers and different compiler versions. We want to avoid using, and especially installing headers that have in them, a macro / identifier / function / variable that conflicts with other projects or the host OS.

That said, we seem to be OK right now with no conflicting macro / identifier / function / variable in builds or the installed headers. Hence, if someone else wants to tackle this, please do so. Otherwise, let's keep the issue around to remind us sometime down the road that this is a desirable change & we'll hopefully find time to do something about it then in the future.