gnuradio/volk #195: let cmake prefer Py3k for builds

jdemel, 2018-07-31:
Since the latest VOLK version builds with Py3k, I suggest to make it the preferred Python version for `cmake`.

There are some advantages
- `Mako` seems to be installed by default on Ubuntu 18.04. Thus, it cuts dependency installs.
- Py3k was identified to be the future. So let's try to make the latest software the default and use everything else as a fallback. This might also be beneficial to flesh out bugs.

How do we get there? Let `cmake` prefer Py3k over Py2k.

noc0lour, 2018-07-31:
We should definitively tell our downstream about that.

michaelld, 2019-11-14:
I'd be game for this change. Doing so keeps Volk better aligned with GNU Radio changes in this regard, beyond other benefits.

jdemel, 2019-12-14:
This issue is fixed with #306. Thus, I close this issue.

closed 2019-12-14 by jdemel