gnuradio/volk #205: NEON broken on raspbian

xloem, 2018-10-08:
It appears that after e98e9277409cd658f1a5b65ffaa0caab842bebce raspbian builds no longer include neon support.

It seems the cmake neon test is run without passing `-mfpu=neon` to the compiler, so it fails:
```
Performing C SOURCE FILE Test neon_compile_result failed with the following output:
Change Dir: /home/pi/src/volk/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_edc61/fast"
/usr/bin/make -f CMakeFiles/cmTC_edc61.dir/build.make CMakeFiles/cmTC_edc61.dir/build
make[1]: Entering directory '/home/pi/src/volk/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_edc61.dir/src.c.o
/usr/bin/cc    -Wall -Dneon_compile_result   -o CMakeFiles/cmTC_edc61.dir/src.c.o   -c /home/pi/src/volk/build/CMakeFiles/CMakeTmp/src.c
/home/pi/src/volk/build/CMakeFiles/CMakeTmp/src.c: In function ‘main’:
/home/pi/src/volk/build/CMakeFiles/CMakeTmp/src.c:2:43: warning: ‘dest’ is used uninitialized in this function [-Wuninitialized]
 int main(){ uint8_t *dest; uint8x8_t res; vst1_u8(dest, res); }
                                           ^~~~~~~~~~~~~~~~~~
/home/pi/src/volk/build/CMakeFiles/CMakeTmp/src.c:2:22: note: ‘dest’ was declared here
 int main(){ uint8_t *dest; uint8x8_t res; vst1_u8(dest, res); }
                      ^~~~
/home/pi/src/volk/build/CMakeFiles/CMakeTmp/src.c:2:43: warning: ‘res’ is used uninitialized in this function [-Wuninitialized]
 int main(){ uint8_t *dest; uint8x8_t res; vst1_u8(dest, res); }
                                           ^~~~~~~~~~~~~~~~~~
/home/pi/src/volk/build/CMakeFiles/CMakeTmp/src.c:2:38: note: ‘res’ was declared here
 int main(){ uint8_t *dest; uint8x8_t res; vst1_u8(dest, res); }
                                      ^~~
In file included from /home/pi/src/volk/build/CMakeFiles/CMakeTmp/src.c:1:0:
/usr/lib/gcc/arm-linux-gnueabihf/6/include/arm_neon.h:9440:1: error: inlining failed in call to always_inline ‘vst1_u8’: target specific option mismatch
 vst1_u8 (uint8_t * __a, uint8x8_t __b)
 ^~~~~~~
/home/pi/src/volk/build/CMakeFiles/CMakeTmp/src.c:2:43: note: called from here
 int main(){ uint8_t *dest; uint8x8_t res; vst1_u8(dest, res); }
                                           ^~~~~~~~~~~~~~~~~~
CMakeFiles/cmTC_edc61.dir/build.make:65: recipe for target 'CMakeFiles/cmTC_edc61.dir/src.c.o' failed
make[1]: *** [CMakeFiles/cmTC_edc61.dir/src.c.o] Error 1
make[1]: Leaving directory '/home/pi/src/volk/build/CMakeFiles/CMakeTmp'
Makefile:126: recipe for target 'cmTC_edc61/fast' failed
make: *** [cmTC_edc61/fast] Error 2

Source file was:
#include <arm_neon.h>
int main(){ uint8_t *dest; uint8x8_t res; vst1_u8(dest, res); }
```

hcab14, 2018-11-22:
Can you try this:
`cmake -DCMAKE_TOOLCHAIN_FILE=../cmake/Toolchains/arm_cortex_a9_hardfp_native.cmake ../`

xloem, 2018-11-22:
@heab14 , that gets the test pasing!  build fails, but perhaps this is a different issue?

```
[ 45%] Building ASM object lib/CMakeFiles/volk_obj.dir/__/kernels/volk/asm/neon/volk_32f_x2_dot_prod_32f_a_neonasm_opts.s.o
/home/pi/src/volk/kernels/volk/asm/neon/volk_32f_x2_dot_prod_32f_a_neonasm_opts.s: Assembler messages:
/home/pi/src/volk/kernels/volk/asm/neon/volk_32f_x2_dot_prod_32f_a_neonasm_opts.s:46: Error: selected processor does not support `sbfx r11,r1,#2,#1' in ARM mode
lib/CMakeFiles/volk_obj.dir/build.make:1793: recipe for target 'lib/CMakeFiles/volk_obj.dir/__/kernels/volk/asm/neon/volk_32f_x2_dot_prod_32f_a_neonasm_opts.s.o' failed
```

balister, 2018-12-29:
The neon flag is passed in via the Toolchain file. Not all armv7 devices have neon, so you can't set it globally. The toolchain file should be chosen based on specific hardware.

balister, 2018-12-29:
I think the arm ode issue is resolved in another pull request.

ast, 2019-08-11:
Resolved by:

https://github.com/gnuradio/volk/pull/270
https://github.com/gnuradio/volk/pull/271

Can be closed...

michaelld, 2019-10-27:
Closing per @ast 's comment.

closed 2019-10-27 by michaelld