gnuradio/volk #210: Make test failed

bachnguyen93, 2018-10-23:
I ran `make test` for the source code version 1.4 from the repository in Xubuntu and received the following error
> The following tests FAILED:
> 3 - qa_volk_32f_log2_32f (Failed)
> 15 - qa_volk_16ic_magnitude_16i (Failed)
> 95 - qa_volk_32f_s32f_convert_32i (Failed)
> 109 - qa_volk_32f_x2_s32f_interleave_16ic (Failed)
> 120 - qa_volk_32fc_s32f_magnitude_16i (Failed)
> Errors while running CTest
> Makefile:105: recipe for target 'test' failed
> make: *** [test] Error 8

Then when I check the version of volk after a `sudo make install`, it gave me version 1.2.1 that was installed together with gnuradio, not the 1.4 that I installed manually from the source code. Did I miss something?

michaelld, 2018-10-23:
The `make test` results aren't a huge surprise. You can run `ctest -V -R qa_volk_32f_log2_32f` to get more specifics as to what's going on; ditto for the other tests. I see some of these tests fail too, due to the precision tolerance being a little too tight (e.g., to 7 decimal places instead of 4).

michaelld, 2018-10-23:
That's odd about the `volk` version being off. Maybe you have another install somewhere? Like, did you have one from the system package manager as well as the one you just installed, in different PREFIXes? That'd be my first guess.

Did you run `sudo ldconfig` after `sudo make install` ... that is generally required to pick up the new libraries.

noc0lour, 2018-10-23:
Since @n-west merged a PR of mine removing the `safeguards` from some tests to actually have some meaning `make test` fails.

michaelld, 2018-10-23:
Ah OK that makes some sense. I've seen more tests fail recently than in prior years. Which PR is it? I'm curious to look through the changes to see what they do & maybe how to fix things for some tests.

noc0lour, 2018-10-23:
https://github.com/gnuradio/volk/pull/199
for the conversion kernels (float -> int) it's quite straightforward to implement fixes. The log2f thing: not so much.

michaelld, 2018-10-23:
OK yeah those are good changes. I see how they might make testing fail. Not sure a good solution off the top of my head. Hopefully someone with more time than me will work things out.

bachnguyen93, 2018-10-24:
Thanks Michael, I did forget to perform the `sudo ldconfig` to link the library

hcab14, 2018-11-23:
Hi @noc0lour, @n-west 

I had a look at `volk_32f_log2_32f ` and realized that the error for the SIMD version is absolute and not relative:
- in the SIMD version the input values are split into exp,frac (w.r.t. powers of two), where frac is between 1 and 2. Then a polynomial approximation is used for log2(frac)/(frac-1)
- the absolute error of this polynomial approximation is < 1e-5
- as this error is independent of the value of the input the error on the output is absolute

What do you think about extending the test cases by one flag which specifies if the tolerance is relative or absolute?

[image]

jdemel, 2019-11-04:
Did #215 solve this issue? What's the current state?

michaelld, 2019-11-14:
Closing due to lack of response and that we've fixed all of the kernels listed here in the past while. If this issue persists for anyone, please reopen this issue or create a new one & provide the output of `make test` and `ctest` for the failing kernels.

closed 2019-11-14 by michaelld