gnuradio/volk #217: volk_32f_s32f_convert_16i, sometimes zero result when misaligned?

ast, 2018-12-19:
```volk_32f_s32f_convert_16i``` seems to sometimes produce a {0} vector when output vector is misaligned on NEON. I'm a bit in the dark here but it would work perfectly on one Raspberry Pi 3 but produce a lot of {0} vectors on another (but not always).

When I changed the output vector from ```std::vector<int16_t>``` to using ```volk_malloc``` problem disappeared.

noc0lour, 2018-12-19:
looks like this might be a bug in the unaligned kernel.

ast, 2018-12-19:
Note: currently I'm using the version of libvolk that comes with raspbian so it might be out of date. What the best way to check version?

Not sure what other info I can supply. I'd be happy to try to build a test case with guidance.

noc0lour, 2018-12-19:
`dpkg -l libvolk` could work?

ast, 2018-12-19:
```
$ dpkg -l libvolk1.3 
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                                   Version                  Architecture             Description
+++-======================================-========================-========================-==================================================================================
ii  libvolk1.3:armhf                       1.3-1+rpi1               armhf                    vector optimized functions```

noc0lour, 2018-12-19:
ok, so it's the 1.3.1 release version

noc0lour, 2018-12-19:
Hm, there is no neon implementation for that kernel? So the generic impl is failing?

noc0lour, 2018-12-19:
This: http://libvolk.org/platforms.html but needs updating for volk 1.4. @n-west how was that table generated?

ast, 2018-12-20:
Never mind about this, I tracked this bug down to a different source. Sorry about that. Still interested in writing volk kernels for NEON though!

ast, 2018-12-21:
I have started writing kernels. Would really like an updated list of missing NEON kernels!

jdemel, 2019-11-06:
@ast Can we close this issue? Was your issue VOLK related? If this is not the case, we should close this issue here.

ast, 2019-11-06:
We can close. I don't remember what caused it but it was not volk releated. Thanks for the push in merging PRs!

jdemel, 2019-11-06:
@ast  Can you close this issue then, please?

ast, 2019-11-06:
Closing as per discussion above.

closed 2019-11-06 by ast