gnuradio/volk #218: A lot of the kernels would benefit from a stride argument.

ast, 2018-12-19:
This has probably been discussed before but I'm currently writing a lot of custom GNU Radio block using volk and it's something I noticed. I find would often need a version of a kernel with a stride argument, most often 2.

This is quite common in other SIMD libraries.

jdemel, 2019-11-09:
Could you point to an example where strides are used?
Also, could you give an example for a kernel that would benefit and how you'd imagine a stride argument to work?
A stride argument would probably make `load` and `store` more complicated. But besides this shouldn't add more complexity. On the other hand, VOLK is designed to work on large arrays. I would always sort data first and then run all the math on aligned data.
We  should also discuss use cases and how VOLK is intended to be used. Otherwise we end up with code that doesn't run as fast as it could for a narrower scope. And we need to figure out if we want to restrict VOLK to this narrower scope anyways.

ast, 2019-11-09:
I have forgotten why I wrote this, I'm sure I had some good reason but let's close it for now.

It probably had something to do with separating complex numbers...

closed 2019-11-09 by ast

michaelld, 2019-11-09:
Strides are used in various algorithm implementations. I've used them before & wondered why we don't have any in Volk. That said, if someone needs them, we can work on adding them.