gnuradio/volk #219: volk_32u_reverse_32u.h: NEON uses `asm` instead of `__VOLK_ASM`

marcusmueller, 2018-12-19:
which leads to compilation issues, gnuradio/gnuradio#2192.

Potential solution: replace `asm` with `__VOLK_ASM`. Can't test.

hcab14, 2018-12-20:
replacing `asm` with `__VOLK_ASM` and not explicitly using GNU compiler extensions (https://github.com/hcab14/volk/commit/e73179535d621e267e91f8cd2f31d799f3b2ea5f) does work, _i.e._, the CI tests on 32 and on 64-bit ARM architectures are passed (https://travis-ci.org/hcab14/volk)

marcusmueller, 2018-12-23:
@hcab14 awesome! Thank you!

nickoe, 2019-02-18:
@hcab14 has a pull request been made?

hcab14, 2019-02-18:
Yes and it is merged into volk/master, see https://github.com/gnuradio/volk/pull/215

nickoe, 2019-02-18:
@hcab14 ahh, ok, thank you. I just missed it because I could see a PR mentioned, and the volk submodule checked out in my gnuradio dir was not updated...

I guess this issue can be closed now then.

hcab14, 2019-10-28:
AFAIC this issue can be closed .

closed 2019-10-28 by michaelld