gnuradio/volk #220: volk_32f_s32f_convert_32i generic off by one.

ast, 2018-12-20:
Should be -2^31 and +2^31-1. Or have I gone crazy?

https://github.com/gnuradio/volk/blob/b459c6fb2ba1b3190dce9805035632ba176e0fab/kernels/volk/volk_32f_s32f_convert_32i.h#L416

noc0lour, 2019-01-07:
So it should only be off-by-one if the float value is `-2^31`, right?  Probably it's safer to replace it with `INT32_{MIN,MAX}` anyway?

ast, 2019-01-07:
INT{16,32}_{MIN,MAX} would be better.

noc0lour, 2019-01-07:
Why INT16 in this case?

ast, 2019-01-07:
Ah not it this case but in `volk_32f_s32f_convert_16i`.

noc0lour, 2019-01-07:
Sure, I agree.

ast, 2019-08-11:
What's the status of this @noc0lour @balister @marcusmueller.

Should we do a search and replace?

jdemel, 2019-11-06:
This issue just got fixed with PR #298 . I guess we can close it now.

michaelld, 2019-11-06:
Agreed!

closed 2019-11-06 by michaelld