gnuradio/volk #221: Building on Raspberry Pi with clang 7 disables neon support

ast, 2018-12-25:
Why is neon not in available machines? Tests pass.
And why is ```avx512f;avx512cd``` there?

```-- Performing Test neon_compile_result
-- Performing Test neon_compile_result - Success
-- Performing Test have_neonv7_result
-- Performing Test have_neonv7_result - Success
-- Performing Test have_neonv8_result
-- Performing Test have_neonv8_result - Failed
-- CPU is armv7, Overruled arch neonv8
-- ORC support not found, Overruled arch orc
-- Available architectures: generic;abm;popcount;norc;avx512f;avx512cd
-- Available machines: generic````

ast, 2018-12-25:
Digging deeper: I believe this is because volk cmake tests for neon but never actually adds it to the ```available_archs```!?  After adding adding the lines below (around line 271 lib/CMakeLists.txt) it builds correctly and works.

I'm using clang 7.0 on a Raspberr Pi 3.

```
   if (have_neonv7_result)
        OVERRULE_ARCH(neonv8 "CPU is armv7")
        list(APPEND available_archs "neon7") # added this!
        list(APPEND available_archs "neon") # added this!
    endif()

    if (have_neonv8_result)
        OVERRULE_ARCH(neonv7 "CPU is armv8")
        list(APPEND available_archs "neonv8") # added this!
        list(APPEND available_archs "neon") # added this!
    endif()
```

ast, 2018-12-25:
```volk_16i_max_star_horizontal_16i.s``` fails to build with this configuration but the rest works.

volk_profile shows **significant** improvements.

balister, 2018-12-26:
Pretty sure neon is no longer an arch. There are two different flavours now.

balister, 2018-12-26:
AH yes, neon selects intrinsics. Can you build with gcc and compare results?

ast, 2018-12-26:
Tried 3 different combinations. Clang 7.0.0, GCC 8 and GCC 6 (default with Raspbian).

Both GCC adds the correct machines like you suspected. However two assembly file fails to compile.

CC=gcc CXX=g++ cmake -DCMAKE_C_FLAGS='-mfpu=neon -mfloat-abi=hard' ..
````
-- Available architectures: generic;hardfp;neon;neonv7;norc
-- Available machines: generic;neon;neonv7_hardfp
````
```
CC=clang CXX=clang++ cmake -DCMAKE_C_FLAGS='-mfpu=neon -mfloat-abi=hard' ..

-- Available architectures: generic;abm;popcount;norc;avx512f;avx512cd
-- Available machines: generic
```

```
volk_32f_x2_dot_prod_32f_a_neonasm_opts.s_nop
volk_32f_x2_dot_prod_32f_a_neonasm.s_nop

/home/pi/src/volk/kernels/volk/asm/neon/volk_32f_x2_dot_prod_32f_a_neonasm_opts.s: Assembler messages:
/home/pi/src/volk/kernels/volk/asm/neon/volk_32f_x2_dot_prod_32f_a_neonasm_opts.s:46: Error: selected processor does not support `sbfx r11,r1,#2,#1' in ARM mode
```

ast, 2018-12-26:
Wouldn't it be easier to migrate neon to only intrinsic?

ast, 2018-12-26:
Clang is missing here:

```<arch name="neon">
  <flag compiler="gnu">-funsafe-math-optimizations</flag>
  <alignment>16</alignment>
  <check name="has_neon"></check>
</arch>

<arch name="neonv7">
  <flag compiler="gnu">-mfpu=neon</flag>
  <flag compiler="gnu">-funsafe-math-optimizations</flag>
  <alignment>16</alignment>
  <check name="has_neonv7"></check>
</arch>```

And as for avx512 I believe it's missing in the overrule list.

ast, 2018-12-26:
Indeed clang is missing in the definitions here: https://github.com/gnuradio/volk/blob/59611a2844f748af08f7fbceab3fe4b1c1b25eee/gen/archs.xml#L16

After I've added it, it works as expected (just copied the gnu arguments).

```
volk_32f_x2_dot_prod_32f_a_neonasm_opts.s_nop
volk_32f_x2_dot_prod_32f_a_neonasm.s_nop
```

Works fine now in clang but not in gcc (it's related to thumb2/thumb somehow) but I found another bug (and fixed I hope) in:

```volk_16i_max_star_horizontal_16i_a_neonasm.s```

https://github.com/gnuradio/volk/issues/222

balister, 2018-12-27:
I've been carrying locally:
https://github.com/balister/meta-sdr/blob/master/recipes-support/volk/files/0001-Force-thumb-mode.patch

ast, 2018-12-27:
But it seems clang has to be in the XML or neon won't be included at all.. Or are you referring to thumb?

ast, 2019-08-11:
Resolved by
https://github.com/gnuradio/volk/pull/271
https://github.com/gnuradio/volk/pull/271

closed 2019-08-11 by ast